wolfSSL/wolfBoot #12: Question: Does wolfBoot builds on windows?

Aukstkalnis, 2019-05-23:
Hi,

I am trying to build wolfBoot on windows, but it is not straight forward. I needed to change makefile to get this going. But now I am failing on python module  wolfcrypt. Does wolfcrypt is supported on windows?

Aukstkalnis, 2019-05-23:
When I try to to download wolfcrypt module, I get this error:
> pip install wolfcrypt
Collecting wolfcrypt
  Using cached https://files.pythonhosted.org/packages/8a/53/d345e1c8ca5ba46328af080b540ce89a7117f2b69a735b100c9aae0ba3c4/wolfcrypt-3.14.2.post0.tar.gz
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "D:\Temp\pip-install-mm06bnrk\wolfcrypt\setup.py", line 56, in <module>
        install_requirements = _parse_requirements("requirements/prod.txt")
      File "D:\Temp\pip-install-mm06bnrk\wolfcrypt\setup.py", line 50, in _parse_requirements
        raw = pip.req.parse_requirements(
    AttributeError: module 'pip' has no attribute 'req'
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in D:\Temp\pip-install-mm06bnrk\wolfcrypt\

danielinux, 2019-05-23:
Hello @Aukstkalnis - we are preparing a new release (4.x) of wolfcrypt-py with the required modules, that should be available via `pip install` soon.

In the meanwhile, wolfcrypt-py module must be installed manually from [sources](https://github.com/wolfSSL/wolfcrypt-py/), via `python setup.py install`.

Aukstkalnis, 2019-05-23:
I see. Thank you. What about wolfboot building on windows? Are you planing to implement something?

danielinux, 2019-05-23:
I've just checked and everything builds normally on windows (if I generate `ed25519.der` and `src/ed25519_pub_key.c` from a Linux box).

I am wsing Win 10 + Cygwin + arm-none-eabi-2018-q4 + gnu make.

Besides the temporary compatibility problems due to wolfcrypt-py 4.x, everything else seems to be working fine.

An alternative would be using [Microsoft Windows 10's WSL](https://docs.microsoft.com/en-us/windows/wsl/install-win10) for now until wolfcrypt-py becomes available on your platform.

Aukstkalnis, 2019-05-23:
Ok, Thanks again 👍I tried WSL and I get this error:
deiauk@aukstkalnisde:/mnt/d/github/wolfBoot$ make
Selected cipher:      ed25519
Output Private key:   ed25519.der
Output C file:        src/ed25519_pub_key.c
Traceback (most recent call last):
  File "tools/keytools/keygen.py", line 65, in <module>
    ed = ciphers.Ed25519Private.make_key(32)
AttributeError: module 'wolfcrypt.ciphers' has no attribute 'Ed25519Private'
Makefile:161: recipe for target 'ed25519.der' failed
make: *** [ed25519.der] Error 1
Do I miss something? I installed wolfcrypt and wolfssl via pip3.

danielinux, 2019-05-23:
As mentioned earlier, the latest available pip package contains wolfcrypt-py 3.x, and a 4.x package is not yet available via pip (it will be released soon).

For the moment, you can get a working wolfcrypt-py 4.0 by doing the following:

-     remove any old wolfcrypt-py site-local packages (on my system they are in /usr/local/lib/python3.7/dist-packages/)
-     clone a fresh wolfcrypt-py from [here](https://github.com/wolfSSL/wolfcrypt-py/)
-     execute `python3 setup.py install` from your local wolfcrypt-py copy of the repository

Thanks for your patience, this steps will be indeed replaced with a single `pip install wolfcrypt` in the near future.

Aukstkalnis, 2019-05-23:
Got it. Thanks again :) I will try this :)

Aukstkalnis, 2019-05-23:
I tried and no compiling stops heare:
Selected cipher:      ed25519
Output Private key:   ed25519.der
Output C file:        src/ed25519_pub_key.c

Creating file ed25519.der
Creating file src/ed25519_pub_key.c
        [CC-ARM] src/ed25519_pub_key.o
        [LD] wolfboot.elf
        [BIN] wolfboot.bin
: not completely converted0
Makefile:138: recipe for target 'wolfboot-align.bin' failed
make: *** [wolfboot-align.bin] Error 1
And I forgot to mention that I try to build for stm32f4 platform and I did not change anything.

danielinux, 2019-05-23:
I cannot reproduce this. It seems related to operations to combine the bootloader "aligned" .bin (different dd version?). 

However, at this stage you already have all the binaries that can be flashed to the stm32f4. If you are using the the default "target.h":
- wolfboot.bin -> flash to 0x08000000
- test-app/image.bin -> flash to 0x08020000

danielinux, 2019-05-23:
sorry obviously the signed version of the application as in `test-app/image_v1_signed.bin`

Aukstkalnis, 2019-05-24:
I was able to workaround this mystery. For some reason "@cat include/target.h |grep WOLFBOOT_PARTITION_BOOT_ADDRESS | head -1 | sed -e "s/.*[ ]//g" > .wolfboot-offset" was printing to .wolfboot-offset file "0x20000\r\n" and not "0x20000". When printf tries to print in decimal, then it fails... I don't understand why this is happening... And one magic thing is that after printf failing, the value in .wolfboot-offset is successfully changed...

danielinux, 2019-05-24:
Thanks for investigating further. Feel free to suggest a fix if you have one that does not break GNU/Linux or MacOS builds.

Aukstkalnis, 2019-05-29:
Adding  tr -d "\n\r"  instead of head -1 helped :) I am using wsl.

danielinux, 2019-05-29:
Fixed in #13

closed 2019-05-29 by danielinux