wolfSSL/wolfBoot #19: Compile error, can't find <stdlib.h>

rgw5267, 2019-08-21:
[image]

[image]

danielinux, 2019-08-26:
Hello @rgw5267 - thanks for reporting this.

I cannot reproduce with my compiler, however I've committed a possible fix in f6727524c414c139212b474af36e029b344e1191 - could you please tell if this fixes the issue for you?

Thanks,

-d

rgw5267, 2019-08-26:
Hi @danielinux 
Thanks for the reply.
I checked out f672752 and the compiler report the same error.
I installed `arm-none-eabi-newlib` and the issue disappered. [https://askubuntu.com/questions/502815/should-arm-none-eabi-gcc-include-a-stdio-h](url)

closed 2019-08-26 by rgw5267