wolfSSL/wolfBoot #66: Compilation error trying to use Keil uVision ARMCC Compiler

ProyMonit, 2020-06-29:
Hello, I´m trying to use WolfBoot in one of my projects. I use Keil uVision 5.27 and the target is STM32F429ZIT6 microcontroller (Nucleo-144 STM32F429 Board). I´m facing the following issue:

Inside boot_arm.c file the function:

```
void RAMFUNCTION do_boot(const uint32_t *app_offset)
{
    mpu_off();
#ifndef NO_VTOR
    /* Disable interrupts */
    asm volatile("cpsid i");
    /* Update IV */
    VTOR = ((uint32_t)app_offset);
#endif

    /* Get stack pointer, entry point */
    app_end_stack = (*((uint32_t *)(app_offset)));
    app_entry = (void *)(*((uint32_t *)(app_offset + 1)));

    /* Update stack pointer */
    asm volatile("msr msp, %0" ::"r"(app_end_stack));
#ifndef NO_VTOR
    asm volatile("cpsie i");
#endif
    /* Unconditionally jump to app_entry */
    asm volatile("mov pc, %0" ::"r"(app_entry));
}
```

The compiler issues the following error:

> Sources\boot_arm.c(236): error:  #18: expected a ")" 

this error appears on lines:

`asm volatile("msr msp, %0" ::"r"(app_end_stack));`
`asm volatile("mov pc, %0" ::"r"(app_entry));`

Apparently this problem is related to the way of inlining assembly trying to pass some parameter. 

Thanks in advance for your help and thanks for sharing this project. Best regards, Yosmany.

dgarske, 2020-06-29:
Hi @ProyMonit ,

It looks like Keil uses `__asm()` syntax. See: http://www.keil.com/support/man/docs/armcc/armcc_chr1359124246903.htm

Can you try altering this inline assembly to use `__asm()` or `__asm volatile()` and see if it resolves the compiler error?

Thanks,
David Garske, wolfSSL

ProyMonit, 2020-06-30:
Thanks for your fast reply. I will try your suggestion and let you know as soon as I can. Best regards.

danielinux, 2020-07-07:
Hello @ProyMonit - could you confirm that switching to __asm volatile() instruction fixes the error with keil? If this is the case we could prepare a specific fix for that compiler.

Thanks in advance, 

D.

ProyMonit, 2020-07-07:
Hello, sorry the delayed reply. I tried the suggested fix but it didn't work. Instead I searched a bit more and finally used this syntax:

```
__asm volatile
    (
        "msr msp, app_end_stack"
    );
```

This last code segment solved the compilation issue. Thanks a lot for the support. Best regards. Yosmany.

danielinux, 2020-07-07:
@ProyMonit Thanks! I will provide a fix ASAP.

/d

danielinux, 2020-07-13:
Hello @ProyMonit - I'm trying to add support for Keil. What I've done so far is in the following branch:
https://github.com/wolfSSL/wolfBoot/tree/keil

The assembly error seems to be gone when I define `asm` as `__asm` when this compiler is detected. The version I'm using is MDK-Lite 5.31.0.0

However, there are a few things about the Keil microVision compiler that are not clear to me, e.g. how to place a custom interrupt vector and initialize all the memory sections at startup like we do with other compilers (gcc, IAR, etc.).

For now the branch is linking a wolfBoot as it were an high-level embedded application using the CMSIS startup code and the built-in reset service routine. The basic case may be working, but many features and platforms won't be properly supported through Keil, and the security of the bootloader may be compromised if the entire boot process is not under direct control of the bootloader code. 

The proposed workaround to make wolfBoot work with Keil at the moment consists in commenting out the interrupt vector table in `boot_arm.c` and the isr_reset function which is normally used as entry point, but I would prefer to define the memory regions at link time and use the interrupt vector configuration and functions defined in wolfBoot. Please see the aforementioned branch for details.

Do you know instead a proper way to implement custom start-up using this tool? Are you using a custom "scatter file" (.sct) for wolfBoot in order to define the required symbols for start/end data/bss and the END_STACK pointer that is placed right before the IV?

Could you please share the project files and the scatter file that you are currently using to compile wolfBoot in Keil, so that we can include and maintain the support in the future versions?

Thanks!

danielinux, 2020-11-03:
Closing - missing information about the report. @ProyMonit feel free to reopen if you are able to share project/scatter files for Keil.

closed 2020-11-03 by danielinux