wolfSSL/wolfBoot #89: Problem with ECC protected flash on STM32H743

reloZid, 2020-11-20:
Hi

I'm currently trying to integrate wolfBoot on a STM32H743 microcontroller. On this chip the flash is ECC protected. Therefore you cannot write to the same flash word (256 bit) more than once, because the ECC value is calculated and stored the first time you write to it. wolfBoot does so when triggering an update: Once to set the partition magic and once to set the partition state. This then results in a fault when reading because the ECC value is invalid.

What is the proper solution to solve this problem? If there is none, are you planning to add support for this use case?

Thanks!

danielinux, 2020-11-20:
Hi,

I'm aware of the problem and I'm trying to figure out a possible solution. Normally we would use `NVM_WRITEONCE`, but on this target the sector size is too big to keep a copy in RAM.

reloZid, 2020-11-22:
Ok, thanks for the answer. For me it works if I change the location of the flags in `get_trailer_at` / `set_trailer_at` such that each flag resides in its own flash word. This wastes some more flash space for the flags, but I'm not sure whether there is a better solution.

danielinux, 2020-11-23:
>  For me it works if I change the location of the flags in get_trailer_at / set_trailer_at such that each flag resides in its own flash word.

I am not sure about a similar fix, since the value of each flag may still be changed twice during the update, without erasing the entire sector.

The workaround committed in ac73034 ensures that when the flags are modified, the current content of the flags word is cached, and restored into flash after a page erase. Please mind that the last 128KB sector in each partition cannot be used to store the firmware, because it will be erased every time the flags are updated.

Thanks,

-- 
Daniele

PR #92 (2020-11-30): Stm32h7 hal fixes
This fixes #89 and ZD11267
closed 2020-11-30 by dgarske