wolfSSL/wolfBoot #93: make config missing WOLFBOOT_DTS_BOOT_ADDRESS and  WOLFBOOT_DTS_UPDATE_ADDRESS

nkapu, 2020-12-07:
Hi, I encountered a potential problem while building wolfBoot

This occurs if someone uses  make config to walk-through setting up the platform, architecture and partition settings.

Problem: Make config doesn’t ask for WOLFBOOT_DTS_BOOT_ADDRESS and WOLFBOOT_DTS_UPDATE_ADDRESS. Problem appears only if there is MMU define in effect for the target architecture.

Compiler:
````
src/image.c: In function ‘wolfBoot_open_image’:
src/image.c:604:44: error: expected expression before ‘:’ token
  604 |                                            : (void*)WOLFBOOT_DTS_UPDATE_ADDRESS;
      |                                            ^
make: *** [Makefile:166: src/image.o] Error 1
````
This isn’t a problem if user uses config/examples as a template to wolfBoot root as .config

dgarske, 2020-12-07:
Hi @nkapu ,

What target are you building for? If using A53 type target the DTS addresses come from the `wolfboot/include/target.h` file. Those addresses are populated via the built-in make scripts via the `.config`. 

For example the config template `config/examples/zynqmp.config` has the following lines:

```
# DTS (Device Tree)
WOLFBOOT_LOAD_DTS_ADDRESS?=0x11800000
WOLFBOOT_DTS_BOOT_ADDRESS?=0x7B0000
WOLFBOOT_DTS_UPDATE_ADDRESS?=0x39B0000
```

```
cp config/examples/zynqmp.config .config
make clean && make
cat include/target.h
```

```
#ifndef H_TARGETS_TARGET_
#define H_TARGETS_TARGET_

#define WOLFBOOT_SECTOR_SIZE                 0x20000
#define WOLFBOOT_PARTITION_BOOT_ADDRESS      0x800000
#define WOLFBOOT_PARTITION_SIZE              0x2A00000
#define WOLFBOOT_PARTITION_UPDATE_ADDRESS    0x3A00000
#define WOLFBOOT_PARTITION_SWAP_ADDRESS      0x63E0000

/* Load address in RAM for staged OS (update_ram only) */
#define WOLFBOOT_DTS_BOOT_ADDRESS            0x7B0000
#define WOLFBOOT_DTS_UPDATE_ADDRESS          0x39B0000
#define WOLFBOOT_LOAD_ADDRESS                0x10000000
#define WOLFBOOT_LOAD_DTS_ADDRESS            0x11800000

#endif /* !H_TARGETS_TARGET_ */
```

If you are not building using the built-in Makefile system then you'll need to manually update your target.h file.

Let me know if that doesn't help.

Thanks,
David Garske, wolfSSL

nkapu, 2020-12-07:
@dgarske Good point. I was referring to someone starting from a scratch and running `make config` instead of copying the `.config` from examples. In that case `tools/config.mk` is missing those variables from `CONFIG_VARS` and they won't be prompted for and compilation fails with confusing error message (in case of `MMU` being defined. Not necessarily anything to fix per se, just having this issue in the repo history might help with google-fu.

danielinux, 2021-02-05:
Thanks for bringing this to our attention!

A fix is now available in #113 and awaiting final review for merge.

closed 2021-02-05 by danielinux