wolfSSL/wolfBoot #105: Device tree blob (dtb) loading is not checked for integrity?

nkapu, 2021-01-31:
Device tree blob (dtb) on Aarch64 does not use partition to check for hash or signature. See e.g.:

 * https://github.com/wolfSSL/wolfBoot/blob/4706d2f12605be98c986759a4d84a5621222a13b/src/update_ram.c#L130
 * https://github.com/wolfSSL/wolfBoot/blob/4706d2f12605be98c986759a4d84a5621222a13b/src/image.c#L602

Is this on purpose? :)

danielinux, 2021-02-01:
Hi @nkapu thanks for the report. 

This is intentional. The DTS is optional, and considered immutable when present. There is no integrity or authenticity verification by default. If you need to update/verify the DTS we would be happy to help, but it would need the corresponding integration with the signing tool as a second image.

nkapu, 2021-02-01:
Thank you for a quick reply. In that context, current documentation in Targets.md might be slightly out-of-date, since for the Raspi it instructs to sign the DTB and embed it in the image as a signed partition: https://github.com/wolfSSL/wolfBoot/blame/bd87099f104460149bc0fb6ed5881d0e63510338/docs/Targets.md#L627-L637

If it helps I can try updating that part of the doc later on if and when I get a slot to get back to this.

danielinux, 2021-02-02:
Good catch. Could you please confirm that the procedure described in #108 reflects the default behavior?

nkapu, 2021-02-02:
Looks good, makes sequence go through to do_boot(). There is still one extra line left https://github.com/wolfSSL/wolfBoot/blame/291b28c430077a9f0fab1fbec0f3bba58f5f9fa3/docs/Targets.md#L630 (unnecessary signing of dtb)

danielinux, 2021-02-03:
Fixed, thanks!

closed 2021-02-03 by danielinux