wolfSSL/wolfBoot #114: Cannot compile IAR example project

kaizoku-oh, 2021-02-05:
The IAR example project for stm32f4 found under IDE/IAR is not able to compile.
After generating the keypair using the generate_key.bat script I wasn't able to compile the project.
I'm using IAR 8.50.9 on Windows 10 x64.

[image]
[image]

danielinux, 2021-02-05:
Hello @kaizoku-oh,

what is the target platform? You may need to define your partitioning in `include/target.h`. When `make` is used, the file is created automatically. In your case, you may want to create it manually starting from `include/target.h.in`, and specifying the flash geometry.

The `include/target.h` automatically generated when using make, configured for the STM32F407 with the parameters in the documentation is the following:

```
#ifndef H_TARGETS_TARGET_
#define H_TARGETS_TARGET_

#define WOLFBOOT_SECTOR_SIZE                 0x20000
#define WOLFBOOT_PARTITION_BOOT_ADDRESS      0x20000
#define WOLFBOOT_PARTITION_SIZE              0x20000 
#define WOLFBOOT_PARTITION_UPDATE_ADDRESS    0x40000
#define WOLFBOOT_PARTITION_SWAP_ADDRESS      0x60000

#endif /* !H_TARGETS_TARGET_ */
```

kaizoku-oh, 2021-02-05:
Yes, that is what I was missing, I created an **include/target.h** file and it compiled without errors (although there were some warnings)
Thanks @danielinux

closed 2021-02-05 by kaizoku-oh