wolfSSL/wolfBoot #120: gicv2_init_secure() crash issue with both Raspberry Pi 3b hardware and Qemu

nkapu, 2021-04-22:
Both Raspberry Pi 3b hardware and Qemu crash in gicv2_init_secure() function https://github.com/wolfSSL/wolfBoot/blob/master/src/boot_aarch64.c#L74. You can get both Qemu and Raspberry Pi to boot linux kernel if it is conditionally (or commenting) out (and an unrelated alignment fix is made) . Seems like gicv2_init_secure() function was originally added for QNX kernel boot on ZynQ https://github.com/wolfSSL/wolfBoot/commit/6b6d5611a7be3e6a56d8325e9a46d56ec45904ee

dgarske, 2021-04-22:
Thanks @nkapu for the report.

It looks like I added that for wolfBoot ZynqMP support with QNX. The QNX platform required initialization of the GIC 400 (GICv2). The Xilinx part here is a Cortex A53. I will note adding a compile time check to only use this for Xilinx. Can you also provide some details on the unrelated alignment fix?
Thanks,
David Garske, wolfSSL

nkapu, 2021-04-29:
Hi @dgarske,

Thanks. I created a separate github issue about the unrelated alignment issue. #122

dgarske, 2022-07-15:
This issue is possibly related to #217

danielinux, 2022-07-25:
Fixed in #223 - closing

closed 2022-07-25 by danielinux