wolfSSL/wolfBoot #208: Core lock up on STM32L0

cmcquinn, 2022-06-07:
I am trying to run the test app for the STM32L0 on a Nucleo-L053R8 and am experiencing a core lock up, before the application even gets into the `isr_reset` function. Any ideas on how to fix this?

danielinux, 2022-06-08:
Hello @cmcquinn could you please share your .config? Are you booting the test app or a custom firmware?

cmcquinn, 2022-06-08:
Hi @danielinux, I am using the example stm32l0 config from `config/examples/stm32l0.config` with the stm32l0 test app from the `test-app` folder.

danielinux, 2022-06-08:
Hello,

The STM32L053 is a very resource-restricted target. The configuration & linker scripts had been tested on the L073 which has sligthly more RAM/FLASH available.

In order to run the demo on the L053 successfully, the following changes are needed:

1) `hal/stm32l0.ld` -> set the memory limit to 8KB (RAM available), by changing the following line:

```
-    RAM (rwx) : ORIGIN = 0x20000000, LENGTH = 20K
+    RAM (rwx) : ORIGIN = 0x20000000, LENGTH = 8K
```

2) Do the same in the application linker script `test-app/ARM.ld` so that the stack pointer will be set to the top of the memory when staging:

```
-    RAM (rwx) : ORIGIN = 0x20000000, LENGTH = 20K
+    RAM (rwx) : ORIGIN = 0x20000000, LENGTH = 8K
```

3) Due to the flash size of 64KB, the configuration made for the L073 will overflow. I suggest a different partition geometry:

```
--- config/examples/stm32l0.config	2022-04-27 14:57:43.000000000 +0200
+++ .config	2022-06-08 19:34:53.147841228 +0200
@@ -16,8 +16,8 @@
 SPMATH?=1
 RAM_CODE?=0
 DUALBANK_SWAP?=0
-WOLFBOOT_PARTITION_SIZE?=0x10000
+WOLFBOOT_PARTITION_SIZE?=0x4000
 WOLFBOOT_SECTOR_SIZE?=0x1000
-WOLFBOOT_PARTITION_BOOT_ADDRESS?=0x8000
-WOLFBOOT_PARTITION_UPDATE_ADDRESS?=0x18000
-WOLFBOOT_PARTITION_SWAP_ADDRESS?=0x28000
+WOLFBOOT_PARTITION_BOOT_ADDRESS?=0x00006000
+WOLFBOOT_PARTITION_UPDATE_ADDRESS?=0x0000a000
+WOLFBOOT_PARTITION_SWAP_ADDRESS?=0x000e000
```

With these three changes I can see my L053 booting into the test application (and turning on the green led).

Thanks,

-- 
Daniele

cmcquinn, 2022-06-24:
This worked. Thanks for your help!

closed 2022-06-24 by cmcquinn