wolfSSL/wolfBoot #209: STM32U5 USART drivers

imranmeah, 2022-06-10:
Is there a plan to develop USART driver support for STM32U5?  I didn't see any USART interfaces in (https://github.com/wolfSSL/wolfBoot/tree/master/hal)/stm32u5.c

I didn't see any STM32U5 support in (https://github.com/wolfSSL/wolfBoot)/[hal](https://github.com/wolfSSL/wolfBoot/tree/master/hal)/uart/

dgarske, 2022-06-10:
Hi @imranmeah ,

No immediate plans, but it is likely 90% copy/paste from one of the L0 or WB55 drivers here: 
https://github.com/wolfSSL/wolfBoot/tree/master/hal/uart
Perhaps @danielinux will be able to provide some pointers?

If you do decide to try implementing it yourself please let us know. We do accept code contributions with a signed agreement. You are welcome to put up a public PR either way.

Thanks,
David Garske, wolfSSL

danielinux, 2022-06-10:
As David said, the controller is perhaps similar to one of those supported in the STM32 family, except perhaps for registers and pin configuration.

Will you perhaps share more details about your project? Are you planning to update the firmware on the target from a neighbour device via USART? Which of the ports available will be likely used? Is this related to a commercial embedded system?

Thanks!

Daniele, wolfSSL

imranmeah, 2022-06-22:
I started to implement the STM32U5 library. Noticed the processor seems to freeze on HAL_UART_Receive(UART_HandleTypeDef *huart, uint8_t *pData, uint16_t Size, uint32_t Timeout). 

The issue is specifically in UART_WaitOnFlagUntilTimeout(huart, UART_FLAG_RXNE, RESET, tickstart, Timeout) != HAL_OK)
It seems to be permanently waiting. The timeout does not happen because system ticks are not triggering. 

 if (((HAL_GetTick() - Tickstart) > Timeout) || (Timeout == 0U))
HAL_GetTick() is returning 0 everytime. 
This function is supposed to return uwTick which In the default implementation, this variable is incremented each 1msin Systick ISR.

Is the interrupt disabled in wolfboot?

dgarske, 2022-06-22:
Hi @imranmeah ,

That is great progress. I am surprised the non interrupt version of the STM HAL API requires the interrupt.

You'll need to map the interrupt function in the vector table here:
https://github.com/wolfSSL/wolfBoot/blob/master/src/boot_arm.c#L435

Thanks,
David Garske, wolfSSL

danielinux, 2023-01-30:
We do have a working UART driver for STM32U5 now, so I'll close this. @imranmeah let us know if you want to suggest any improvements based on the latest master branch.

Thanks!

-- 
Daniele - wolfSSL

closed 2023-01-30 by danielinux