wolfSSL/wolfBoot #224: IAR linking errors

kaizoku-oh, 2022-07-22:
The functions  `keystore_get_xxx()` definitions are not generated by the keygen.exe tool
I'm just using the generate_key.bat with the following content and it generates the public and private keys as expected but doesn't add these functions:

```batch
keytools\keygen.exe --ecc256 ..\src\bootloader\src\ecc256_pub_key.c
move ecc256.der ..\src\bootloader
```

[image]

danielinux, 2022-07-22:
hello @kaizoku-oh thanks for reporting this.

These issues with IAR are due to the fact that we changed how keys are created, and the file names for the key files.

- old `keygen.exe` was getting the path for the generated .c pubkey, and implying the name for the private key (ecc256.der in your case)
- new `keygen.exe` usage implies the creation of the keystore.c file containing the new public key structure, while the name for the private key is read from the command line (-g argument), e.g.: `keygen.exe -g private_key.der`

I'm trying to get a windows machine with IAR to provide some fixes. Thanks for your patience.

kaizoku-oh, 2022-07-25:
Hi @danielinux thanks for your feedback.
While waiting I decided to try the python keygen tool (keygen.py) in WSL and this is what I got.
I'm not sure what's missing.

Also here is my GitHub repo, it contains one IAR workspace with 2 projects: bootloader and application
I'm using IAR v9.30 and ST NUCLEO-F401RE board
https://github.com/kaizoku-oh/wolfboot-iar-example

[image]

danielinux, 2022-07-25:
Hello, it could be that your python version is outdated. The new keygen.py requires python >= v3.8. Please ensure you update python in your VM, or use the C tool.

Thanks,

--
Daniele

danielinux, 2022-07-25:
hi @kaizoku-oh ,

My colleague David has prepared a fix in PR #230 that should fix the problem you reported. The `generate_key.bat` and the precompiled version of `keygen.exe` have been updated as well.

closed 2022-07-26 by danielinux

kaizoku-oh, 2022-07-26:
Thank you very much @danielinux and @dgarske for your support.