wolfSSL/wolfBoot #232: Running WolfBoot on  ST NUCLEO-F401RE

kaizoku-oh, 2022-07-29:
Hello,

I'm trying to run wolfboot on an STM32F401RE along with a user application generated with cubeMX.
here is my repo: https://github.com/kaizoku-oh/wolfboot-iar-example
Basically I have an IAR workspace that contains 2 projects bootloader(wolfboot) and application (blinky + printf).
I also added 3 batch scripts to remove and generate keys as well as sign the application binary.

I can compile both projects (bootloader and application) with no problem.
But after flashing the binaries nothing happens, (if the app starts the led should be blinking and should be able to see UART messages).
I don't know if I'm missing something or if there is a way to debug the bootloader to see why it didn't jump to the application.

Here is the configuration I used and the steps I followed for both bootloader and application:
# Bootloader
**IAR preprocessor defines:**
```c
__WOLFBOOT
PLATFORM_stm32f4
ARCH_ARM
WOLFSSL_USER_SETTINGS
DEBUG_CONSOLE_ASSERT_DISABLE=1
WOLFBOOT_SIGN_ECC256
WOLFBOOT_HASH_SHA256
WOLFSSL_SP_ASM
WOLFSSL_SP_ARM_CORTEX_M_ASM
ARCH_FLASH_OFFSET=0x08000000
WOLFBOOT_FIXED_PARTITIONS
```

**target.h:** (common file used by bootloader and application)
```c
#ifndef H_TARGETS_TARGET_
#define H_TARGETS_TARGET_

#define WOLFBOOT_SECTOR_SIZE                 0x20000
#define WOLFBOOT_PARTITION_BOOT_ADDRESS      0x20000
#define WOLFBOOT_PARTITION_SIZE              0x20000 
#define WOLFBOOT_PARTITION_UPDATE_ADDRESS    0x40000
#define WOLFBOOT_PARTITION_SWAP_ADDRESS      0x60000

#endif /* !H_TARGETS_TARGET_ */
```

**1. generate keys:** [generate_key.bat](https://github.com/kaizoku-oh/wolfboot-iar-example/blob/main/tools/generate_key.bat)
```ps
PS D:\wolfboot-iar-example\tools> .\generate_key.bat
```
**2. build the bootloader project**

# Application
**IAR preprocessor defines:**
```c
__WOLFBOOT
USE_HAL_DRIVER
STM32F401xE
WOLFBOOT_FIXED_PARTITIONS
PLATFORM_stm32f4
ARCH_ARM
WOLFSSL_USER_SETTINGS
DEBUG_CONSOLE_ASSERT_DISABLE=1
WOLFBOOT_SIGN_ECC256
WOLFBOOT_HASH_SHA256
WOLFSSL_SP_ASM
WOLFSSL_SP_ARM_CORTEX_M_ASM
```

**IAR ICF linker file:** (as you can see I modified the flash start address from **0x08000000** to **0x08020100**
```icf
/*###ICF### Section handled by ICF editor, don't touch! ****/
/*-Editor annotation file-*/
/* IcfEditorFile="$TOOLKIT_DIR$\config\ide\IcfEditor\cortex_v1_0.xml" */
/*-Specials-*/
define symbol __ICFEDIT_intvec_start__ = 0x08020100;
/*-Memory Regions-*/
define symbol __ICFEDIT_region_ROM_start__    = 0x08020100;
define symbol __ICFEDIT_region_ROM_end__      = 0x0803FFFF;
define symbol __ICFEDIT_region_RAM_start__    = 0x20000000;
define symbol __ICFEDIT_region_RAM_end__      = 0x20017FFF;
/*-Sizes-*/
define symbol __ICFEDIT_size_cstack__ = 0x400;
define symbol __ICFEDIT_size_heap__   = 0x200;
/**** End of ICF editor section. ###ICF###*/

define memory mem with size = 4G;
define region ROM_region      = mem:[from __ICFEDIT_region_ROM_start__   to __ICFEDIT_region_ROM_end__];
define region RAM_region      = mem:[from __ICFEDIT_region_RAM_start__   to __ICFEDIT_region_RAM_end__];

define block CSTACK    with alignment = 8, size = __ICFEDIT_size_cstack__   { };
define block HEAP      with alignment = 8, size = __ICFEDIT_size_heap__     { };

initialize by copy { readwrite };
do not initialize  { section .noinit };

place at address mem:__ICFEDIT_intvec_start__ { readonly section .intvec };

place in ROM_region   { readonly };
place in RAM_region   { readwrite,
                        block CSTACK, block HEAP };
```

**system_stm32f4xx.c:**
```c
#define USER_VECT_TAB_ADDRESS

#define VECT_TAB_BASE_ADDRESS   FLASH_BASE      /*!< Vector Table base address field.
                                                     This value must be a multiple of 0x200. */
#define VECT_TAB_OFFSET         0x00020100U     /*!< Vector Table base offset field.
                                                     This value must be a multiple of 0x200. */
```

**1. build the application project**

**2. sign app binary:** [sign_app.bat](https://github.com/kaizoku-oh/wolfboot-iar-example/blob/main/tools/sign_app.bat)

```ps
PS D:\wolfboot-iar-example\tools> .\sign_app.bat 1
```

# Flashing
I followed the same steps mentioned in the docs [here](https://github.com/wolfSSL/wolfBoot/blob/master/IDE/IAR/README.md)

danielinux, 2022-08-01:
Hi @kaizoku-oh, 
It looks like you are using interrupt handlers in the application. 

As per the following comment you pasterd from `system_stm32f4xx.c:`:

```
#define VECT_TAB_OFFSET         0x00020100U     /*!< Vector Table base offset field.
                                                     This value must be a multiple of 0x200. */
```

This would require an entry point that's a multiple of 0x200 for IV to be correctly remapped upon staging.

Could you try changing the entry point for your IV to 0x(08)020200 and adding the following define to your wolfBoot build:

```
IMAGE_HEADER_SIZE=0x200
```

Please also ensure that the sign tool that you are using (c or python) picks up the correct IMAGE_HEADER_SIZE parameter. `sign.py` takes this value from `.config`. If you are using `sign.c` you may want to ensure that the define `IMAGE_HEADER_SIZE=0x200` is provided at compile time.

Please let us knows whether this fixes the problem for you.

Thanks,

-- 
Daniele - wolfSSL

kaizoku-oh, 2022-08-02:
Hi @danielinux thanks for your response, here the changes I made and the steps I followed but I still get the same result: the application is not booting.

# Changes:
**1. Updated the** `VECT_TAB_OFFSET` **in** `system_stm32f4xx.c:`
```c
#define VECT_TAB_OFFSET         0x00020200U     /*!< Vector Table base offset field.
                                                     This value must be a multiple of 0x200. */
```

**2. Updated flash linker file:**

```icf
define symbol __ICFEDIT_region_ROM_start__    = 0x08020200;
```
**3. Added header size define to wolfboot project preprocessor:**
`IMAGE_HEADER_SIZE=0x200`

**4. Replaced `config.mk` file content with the following:**
```mk
#Defaults, used by CI
ifeq ($(ARCH),)
  ARCH?=ARM
  TARGET?=stm32f4
  SIGN?=ED25519
  HASH?=SHA256
  IMAGE_HEADER_SIZE?=512
endif

CONFIG_VARS:= ARCH TARGET SIGN HASH IMAGE_HEADER_SIZE
```

# Steps:
**1. Generate keys**
**2. Build wolfboot with IAR**
**3. Build application with IAR**
**5. Sign the application by executing sign.py from WSL:**
```bash
$ python3.10 src/bootloader/tools/keytools/sign.py --ecc256 --sha256 prj/application/Debug/Exe/application.bin prj/bootloader/wolfboot_signing_private_key.der 1
wolfBoot KeyTools (Python version)
wolfcrypt-py version: 5.4.0-0
Update type:          Firmware
Input image:          prj/application/Debug/Exe/application.bin
Selected cipher:      ecc256
Private key:          prj/bootloader/wolfboot_signing_private_key.der
Output image:         prj/application/Debug/Exe/application_v1_signed.bin
Not Encrypted
Target partition id:  1
Calculating sha256 digest...
Signing the firmware...
Done.
Output image successfully created.
$
```
**6. Erase all flash memory**
**7. Flash bootloader at** `0x08000000`
**8. Flash signed application at** `0x08020000` (skip flash erase)

danielinux, 2022-08-02:
> src/bootloader/tools/keytools/sign.py --ecc256 --sha256 prj/application/Debug/Exe/application.bin prj/bootloader/wolfboot_signing_private_key.der 1

- problem 1:I don't see the header size here. You should create a .config file containing the Iine IMAGE_HEADER_SIZE=512. When the sign.py reads the manifest header size from the file, it should print it, like the following:

```
wolfcrypt-py version: 5.3.0-0
IMAGE_HEADER_SIZE (from .config): 512
Update type:          Firmware
Input image:          test-app/image.bin
Selected cipher:      ecc256
Private key:          wolfboot_signing_private_key.der
Output image:         test-app/image_v1_signed.bin
Not Encrypted
Target partition id:  1 
Calculating sha256 digest...
Signing the firmware...
Done.
Output image successfully created.
```

- problem 2: There is a mismatch with the algorithms selected. In your `config.mk` you are picking ED25519, then you are signing the image with a different algorithm (ecc256). Please use the same algo for both.

- note: if you decide to use ed25519, please downgrade your wolfcrypt-py to 5.3.0.post0 - there is a known issue with the latest version with ED25519 that won't allow you to import keys properly. ECC256 is fine on this version. To downgrade wolfcrypt-py for usage with ED25519, use:

`pip install wolfcrypt==5.3.0.post0`

kaizoku-oh, 2022-08-02:
**problem 1:**
Actually I modified the existing `config.mk` file with the following but it outputs the following (without **IMAGE_HEADER_SIZE (from .config): 512**)
```mk
#Defaults, used by CI
ifeq ($(ARCH),)
  ARCH?=ARM
  TARGET?=stm32f4
  SIGN?=ECC256
  HASH?=SHA256
  IMAGE_HEADER_SIZE?=512
endif

CONFIG_VARS:= ARCH TARGET SIGN HASH IMAGE_HEADER_SIZE

```

danielinux, 2022-08-02:
you can verify `sign.py` is successful if you open the signed file `prj/application/Debug/Exe/application_v1_signed.bin` with a hex editor and see the manifest and 0xFF padding until `0x0200`. You can modify the sign.py script manually to set `IMAGE_HEADER_SIZE` otherwise.

danielinux, 2022-08-02:
unfortunately, the choice of using IAR for wolfboot leads to a few things being harder to customize (like the IMAGE_HEADER_SIZE here). Using the default GNU toolchain is definitely recommended in general. Application and bootloader can still be compiled with different toolchains as well, e.g. if you create a `.config` file for your platform and run `make wolfboot.bin` from wolfboot root directory

kaizoku-oh, 2022-08-02:
**I get it now (the sign.py + .config):**
you need to put the `.config` in the same directory from you execute the `python sign.py` command
It would make sense if the sign.py can accept the .config path as argument.

danielinux, 2022-08-02:
Thanks for the suggestion, we'll definitely consider that.

kaizoku-oh, 2022-08-02:
Is there a way to debug the bootloader and find out why it can't jump to the app?

danielinux, 2022-08-02:
Yes, that header looks fine. It would be appreciated if you could compile wolfboot with debugging symbols, and step through `wolfBoot_start` to check in particular if the application can be verified, or if you get to `wolfBoot_panic` because the signature is invalid or the SHA does not match. If the function wolfBoot_start calls `wolfBoot_panic()` you can see the results of the verifications by printing the members of the `boot` structure, of type `wolfBoot_image`. It contains a flag for each verification.

A breakpoint in `do_boot`, if hit, means that wolfBoot verified the application correctly, and there is something going wrong in the application bring-up.

kaizoku-oh, 2022-08-02:
An BusFault exception happens in main() --> hal_init() --> clock_pll_on()

[image]

danielinux, 2022-08-02:
The PLL parameters and CPU clock in the hal driver that you are using are set for STM32F407, running at 168MHz with 8MHz crystal. In order to run wolfBoot you will have to adjust the parameters in the `clock_pll_on` function to match your hardware (CPU clock speed, source clock configuration and oscillator mounted on board). I believe ST provides a tool to calculate the pll parameters and the prescaler divider factors for STM32 boards. You can safely comment out the calls to clock_pll_on and clock_pll_off from hal_init/hal_prepare_boot functions. The board will run much slower and it will take several seconds (or even minutes, depending on the default clock speed) to verify the image.

I will close this issue for now. If you need assistance to complete the port to your specific target board let's continue this conversation through support, please write to us at support at wolf ssl dot com,

Thanks,

-- 
Daniele

closed 2022-08-02 by danielinux