wolfSSL/wolfBoot #234: `master` does not build with `wolfcrypt` 5.4.0

cmcquinn, 2022-08-15:
The `master` branch does not build with `wolfcrypt` 5.4.0. Build returns the following error:

```
Traceback (most recent call last):
  File "/home/cameron/wolfBoot/tools/keytools/sign.py", line 562, in <module>
    ed = ciphers.Ed25519Private(key = wolfboot_key_buffer)
  File "/home/cameron/.local/lib/python3.10/site-packages/wolfcrypt/ciphers.py", line 1233, in __init__
    self.decode_key(key)
  File "/home/cameron/.local/lib/python3.10/site-packages/wolfcrypt/ciphers.py", line 1271, in decode_key
    raise WolfCryptError("Key decode error (%d)" % ret)
wolfcrypt.exceptions.WolfCryptError: Key decode error (-173)
```

`wolfcrypt` was installed using `pip` from PyPI. This error occurred using both the sim and stm32l0 examle configs.

kareem-wolfssl, 2022-08-15:
I'm able to reproduce this issue here.
As a temporary workaround you can use our native keygen tool instead:

```
make keygen
make
```

I'm working on debugging why our Python tool is failing and will update when I have more information.

danielinux, 2022-08-16:
Hi, this is a known issue of wolfcrypt-py 5.4.0 release when using ED25519.

I'd suggest to revert to wolfcrypt-py 5.3.0 if you want to use ED25519.

`pip install wolfcrypt==5.3.0.post0` fixes the issue.

Thanks,

-- 
Daniele

closed 2022-08-16 by danielinux