wolfSSL/wolfBoot #241: Compilation for cortex-M0

Thisora, 2022-10-12:
I'm currently trying to use wolfboot as stage 3 bootloader for RP2040. 
I made HAL and Linker script.
When i try to compile WolfBoot, i have the following error:
....

[build] Output image(s) successfully created.
[build] lib/wolfssl/wolfcrypt/src/sp_armthumb.c: In function 'sp_256_mul_8':
[build] lib/wolfssl/wolfcrypt/src/sp_armthumb.c:96958:1: error: r7 cannot be used in 'asm' here
[build] 96958 | }
[build]       | ^
[build] make[3]: *** [lib/wolfssl/wolfcrypt/src/sp_armthumb.o] Error 1
[build] make[2]: *** [CMakeFiles/wolfboot] Error 2
[build] make[1]: *** [CMakeFiles/wolfboot.dir/all] Error 2

This come from the configuration CORTEX_M0?=1 
(because rp2040 has dual-core Cortex-M0+ )
If i remove it (Cortex-M0=0) , it compile correctly.

Am i doing something wrong ?

danielinux, 2022-10-12:
Hi!
Can you paste the entire compile line (like with `make V=1`)? To use sp_armthumb you need to set a few CPU flags in gcc. Those should be automatically added by arch.mk when compiling on M0 (`-mthumb-interwork -mcpu=cortex-m0`). 

Interesting that you are porting to this target. Could you tell us more about this project? Do you want to share your hal/ld for inclusion? What are your plans to store the public key on the rpi-pico?

Thanks,

-- 
Daniele

Thisora, 2022-10-12:
Hi !
Thank you for your answer !
Oh my bad i had to use V=1 i modified my .config file and it seems to work right now.

I'm working on my master thesis where i have to create an update system for a bare metal application that is running on RP2040. I look for a bootloader that is able to provide an A/B seamless update system and maybe there will be a TPM to store keys. That's why wolfboot seems perfect. For now i just to make sure that i can get it working on RP2040.

Yes of course i'll share hal/ld if i can get it working. But as i'am going to use it for stage 3 the linker script is going to be a bit specific to this project. But if i can get it i will of course share with you my work if it interest you.

Thank you for your help

Thisora, 2022-10-13:
Tested again this morning and it seems like the error message came back. I don't know what did i changed.

Here is my config file: 

```
ARCH?=ARM
TARGET?=pico
SIGN?=ECC256
HASH?=SHA256
DEBUG?=1
VTOR?=1
CORTEX_M0?=1
NO_ASM?=0
EXT_FLASH?=0
SPI_FLASH?=0
ALLOW_DOWNGRADE?=0
NVM_FLASH_WRITEONCE?=0
WOLFBOOT_VERSION?=1
V?=1
SPMATH?=1
RAM_CODE?=1
WOLFBOOT_PARTITION_SIZE?=0x60000
WOLFBOOT_SECTOR_SIZE?=0x20000
WOLFBOOT_PARTITION_BOOT_ADDRESS?=0x1000FA00
WOLFBOOT_PARTITION_UPDATE_ADDRESS?=0x1006FA00
WOLFBOOT_PARTITION_SWAP_ADDRESS?=0x100CFA00
```

And here is the compilation line and the error message:

```
        [CC-ARM] lib/wolfssl/wolfcrypt/src/sp_armthumb.o
arm-none-eabi-gcc -D"__WOLFBOOT" -Werror -Wextra -mthumb -mlittle-endian -mthumb-interwork -DARCH_ARM -mcpu=cortex-m0 -DWOLFSSL_SP_ASM -DWOLFSSL_SP_ARM_THUMB_ASM -DARCH_FLASH_OFFSET=0x0 -D"WOLFBOOT_SIGN_ECC256" "-Wstack-usage=5880" -D"RAM_CODE" -O0 -g -ggdb3 -DWOLFBOOT_VERSION=1 -D"WOLFBOOT_HASH_SHA256" -DIMAGE_HEADER_SIZE=256 -I"." -I"include/" -I"lib/wolfssl" -D"WOLFSSL_USER_SETTINGS" -D"WOLFTPM_USER_SETTINGS" -D"PLATFORM_pico" -Wall -Wextra -Wno-main -ffreestanding -Wno-unused -nostartfiles -ffunction-sections -fdata-sections -c -o lib/wolfssl/wolfcrypt/src/sp_armthumb.o lib/wolfssl/wolfcrypt/src/sp_armthumb.c
lib/wolfssl/wolfcrypt/src/sp_armthumb.c: In function 'sp_256_mul_8':
lib/wolfssl/wolfcrypt/src/sp_armthumb.c:96958:1: error: r7 cannot be used in 'asm' here
96958 | }
      | ^
make: *** [lib/wolfssl/wolfcrypt/src/sp_armthumb.o] Error 1
```

danielinux, 2022-10-13:
Hello again. Can you please provide a few more details? This would help me reproduce the problem, although I have the impression that this issue has been recently fixed in wolfcrypt.

- wolfBoot version/hash (perhaps via `git show` if you cloned the repo from here)
- wolfSSL version/hash (`git show` in lib/wolfssl)
- compiler version (`arm-none-eabi-gcc -v`)

Thanks,

-- 
Daniele - wolfSSL

Thisora, 2022-10-13:
Commit hash : c346fab41b98ffd14ec18bbf7b440bb3e0c95991
Wolfssl version: 5.5.1
Compiler version: 10.3-2021.07

danielinux, 2022-10-13:
I've matched your versions but I'm still unable to reproduce. Are you sure about the wolfSSL version? Is that not coming from git? Did you install wolfSSL manually in lib/wolfssl instead of running `git submodule update`? If this is the case, could you try deleting the directory lib/wolfssl and running `git submodule update`

My successful command line for that file:

```
[CC-ARM] lib/wolfssl/wolfcrypt/src/sp_armthumb.o
/usr/bin/arm-none-eabi-gcc -D"__WOLFBOOT" -Werror -Wextra -mthumb -mlittle-endian -mthumb-interwork -DARCH_ARM -mcpu=cortex-m0plus -DWOLFSSL_SP_ASM -DWOLFSSL_SP_ARM_THUMB_ASM -DARCH_FLASH_OFFSET=0x0 -D"WOLFBOOT_SIGN_ECC256" "-Wstack-usage=5880" -D"RAM_CODE" -D"WOLFBOOT_FLAGS_INVERT=1" -D"NVM_FLASH_WRITEONCE" -Os -D"WOLFBOOT_HASH_SHA256" -DIMAGE_HEADER_SIZE=256 -I"." -I"include/" -I"lib/wolfssl" -D"WOLFSSL_USER_SETTINGS" -D"WOLFTPM_USER_SETTINGS" -D"PLATFORM_stm32l0" -Wall -Wextra -Wno-main -ffreestanding -Wno-unused -nostartfiles -ffunction-sections -fdata-sections -c -o lib/wolfssl/wolfcrypt/src/sp_armthumb.o lib/wolfssl/wolfcrypt/src/sp_armthumb.c
```
Thanks,

-- 
Daniele

Thisora, 2022-10-13:
Okey i tried your command and work fine.
I tried with every same parameters except the  plateforme and it make the same error.

To resume:
It work with `-D"PLATFORM_stm32l0"`
but give error with `-D"PLATFORM_pico"`

The fact is plateform_pico is what i've added for RP2040.

Maybe i missconfigured something ?
I've followed the instruction to port on new plateforme (create HAL/ld). I have to modify a little bit the Makefile but only to set my app path (../build/image.elf instead of test-app/image.elf) and some minore change that may not impact the compilation of wolfssl.

(I think the first time i closed the issue was because i tested with option Cortex-M0?=0 and i frogot to `make clean` to test with Cortex-M0?=1).

Thank you for your help !

danielinux, 2022-10-13:
It looks like the problem is with the -O0. I will ask if this is a known issue with sp_armthumb and optimization levels. 
Any other optimization level (-Os, -O1, -O2) seem to work fine.

I can now reproduce with `make DEBUG=1`

Solution: disable debug with your build, or use `NO_ASM=1` as long as you want debugging symbols and `-O0` in place.

Thanks for spotting this. I will report internally.

Thisora, 2022-10-13:
Yes, it works !
Thank you !

closed 2022-10-13 by Thisora