wolfSSL/wolfBoot #244: Reset keystore.der

Thisora, 2022-10-19:
Hi !
I finally got this working and was able to boot.
I try differents things to test the wolfboot but when i tried to generate new keys (`make keysclean` `make keys`). I was not able to boot anymore. I found with debugger that the authenticity check fail right there:
```c
static int keyslot_id_by_sha(const uint8_t *hint)
{
    int id = 0;
    for (id = 0; id < keystore_num_pubkeys(); id++)
    {
        key_hash(id, digest);
        if (memcmp(digest, hint, WOLFBOOT_SHA_DIGEST_SIZE) == 0)
            return id;
    }
    return -1;
}
```

Because the `digest` kept the old value (before i regenerate the keystore). Probably i missed something but i can't find anything in documentation.

I really don't know why but i had to modify a lot of Makefile to get working with RP2040. I made a fork if you want add hal/ld for rp2040 but won't work without the stage2 bootloader of pick-sdk. Feel free to contact me if you want more infos.

danielinux, 2022-10-19:
Yes, this is a mismatch between the keys compiled in and the one used to sign the image.

Have you rebuilt wolfboot after updating the keys? `keystore.der` and `src/keystore.c` are updated when new keys are generated, which requires to rebuild wolfboot and sign your image again using the new private key.

Thisora, 2022-10-19:
Yes i did `make clean` `make` and still have same problem.

Thisora, 2022-10-19:
Ok my bad i problem with my flash system which caused to not update image. That's why i had previous digest 🥺.
Thank you for support !

closed 2022-10-19 by Thisora

danielinux, 2022-10-19:
Alright! Glad it's solved.