wolfSSL/wolfBoot #249: Values of `version_field` and `type_field` are ignored by compiler optimization

Nanamiiiii, 2022-11-09:
I'm trying to use wolfBoot on QEMU virt machine model for RISC-V.
When updating firmware from emulated flash, I found that update partition was not measured correctly.

Checking compiled binary, I noticed that the `type_field` and `version_field` values are ignored.
Each variable is used in `libwolfboot.c#wolfBoot_get_image_type()` and `libwolfboot.c#wolfBoot_get_blob_version()`. 

I also find same variable in functions related delta update, but I did not try delta update and don't know whether this has same issue.

## Details

By default, gcc optimization seems to set to `-Os`. This options generate the assembly below.
```asm
0000000080000de2 <wolfBoot_get_blob_version>:
    80000de2:   4118                    lw      a4,0(a0)
    80000de4:   1101                    addi    sp,sp,-32
    80000de6:   464c57b7                lui     a5,0x464c5
    80000dea:   ec06                    sd      ra,24(sp)
    80000dec:   e402                    sd      zero,8(sp)
    80000dee:   f5778793                addi    a5,a5,-169 # 464c4f57 <STACK_SIZE+0x464c4b57>
    80000df2:   00f71763                bne     a4,a5,80000e00 <wolfBoot_get_blob_version+0x1e>
    80000df6:   0030                    addi    a2,sp,8
    80000df8:   4585                    li      a1,1
    80000dfa:   0521                    addi    a0,a0,8
    80000dfc:   f73ff0ef                jal     ra,80000d6e <wolfBoot_find_header>
    80000e00:   60e2                    ld      ra,24(sp)
    80000e02:   4501                    li      a0,0
    80000e04:   6105                    addi    sp,sp,32
    80000e06:   8082                    ret
```
The function expects to return the value of the `version_field`, but 0 is stored directly in `a0` through any flow, and 0 is always returned. I investigated using the debugger and found that this was the cause of the problem.

I fixed this problem by simply adding `volatile` to `version_field` or `type_field`.
This is example of `version_field`.
```c
uint32_t wolfBoot_get_blob_version(uint8_t *blob)
{
    uint32_t * volatile version_field = NULL;
    uint32_t *magic = NULL;

    magic = (uint32_t *)blob;
    if (*magic != WOLFBOOT_MAGIC)
        return 0;
    if (wolfBoot_find_header(blob + IMAGE_HEADER_OFFSET, HDR_VERSION,
            (void *)&version_field) == 0)
        return 0;
    if (version_field)
        return im2n(*version_field);
    return 0;
}
```
Assembly changed to this.
```asm
0000000080000de2 <wolfBoot_get_blob_version>:
    80000de2:   4118                    lw      a4,0(a0)
    80000de4:   1101                    addi    sp,sp,-32
    80000de6:   464c57b7                lui     a5,0x464c5
    80000dea:   ec06                    sd      ra,24(sp)
    80000dec:   e402                    sd      zero,8(sp)
    80000dee:   f5778793                addi    a5,a5,-169 # 464c4f57 <STACK_SIZE+0x464c4b57>
    80000df2:   00f70663                beq     a4,a5,80000dfe <wolfBoot_get_blob_version+0x1c>
    80000df6:   4501                    li      a0,0
    80000df8:   60e2                    ld      ra,24(sp)
    80000dfa:   6105                    addi    sp,sp,32
    80000dfc:   8082                    ret
    80000dfe:   0030                    addi    a2,sp,8
    80000e00:   4585                    li      a1,1
    80000e02:   0521                    addi    a0,a0,8
    80000e04:   f6bff0ef                jal     ra,80000d6e <wolfBoot_find_header>
    80000e08:   d57d                    beqz    a0,80000df6 <wolfBoot_get_blob_version+0x14>
    80000e0a:   67a2                    ld      a5,8(sp)
    80000e0c:   d7ed                    beqz    a5,80000df6 <wolfBoot_get_blob_version+0x14>
    80000e0e:   67a2                    ld      a5,8(sp)
    80000e10:   4388                    lw      a0,0(a5)
    80000e12:   b7dd                    j       80000df8 <wolfBoot_get_blob_version+0x16>
```
It correctly returned value of `version_field`.

I can't determine if this problem is specific to my environment, so I will share it with you.

Compiler toolchain: [RISC-V GNU Toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) Tag:2022.06.10
Options in `arch.mk` is below. Some sources are modified by me for riscv64.
```make
## RISCV
ifeq ($(ARCH),RISCV)
  ifeq ($(TARGET), riscv_virt)
    ## For Qemu virt emulation
    CROSS_COMPILE:=riscv64-unknown-elf-
    CFLAGS+=-fno-builtin-printf -DUSE_M_TIME -g -march=rv64imac -mabi=lp64 -mcmodel=medany -nostartfiles -DARCH_RISCV -DRISCV_QEMU_VIRT
    LDFLAGS+=-march=rv64imac -mabi=lp64 -mcmodel=medany
    MATH_OBJS += ./lib/wolfssl/wolfcrypt/src/sp_c32.o
    CFLAGS +=-ffunction-sections -fdata-sections
    LDFLAGS+=-Wl,--gc-sections
    OBJS+=src/boot_riscv64.o src/vector_riscv64.o
  else
```

Thanks.

danielinux, 2022-11-10:
Hello,

Thanks for reporting this. I'm going to run a few tests to confirm and test the proposed solution.

danielinux, 2022-11-28:
I can confirm that this happens on other architectures as well.

On ARM Cortex-M I get:
```
000007dc <wolfBoot_get_blob_version>:
     7dc:   2300        movs    r3, #0
     7de:   b507        push    {r0, r1, r2, lr}
     7e0:   6802        ldr r2, [r0, #0]
     7e2:   9301        str r3, [sp, #4] 
     7e4:   4b05        ldr r3, [pc, #20]   ; (7fc <wolfBoot_get_blob_version+0x20>)
     7e6:   429a        cmp r2, r3
     7e8:   d104        bne.n   7f4 <wolfBoot_get_blob_version+0x18>
     7ea:   2101        movs    r1, #1
     7ec:   aa01        add r2, sp, #4
     7ee:   3008        adds    r0, #8
     7f0:   f7ff ffcc   bl  78c <wolfBoot_find_header>
     7f4:   2000        movs    r0, #0
     7f6:   b003        add sp, #12 
     7f8:   f85d fb04   ldr.w   pc, [sp], #4
     7fc:   464c4f57            ; <UNDEFINED> instruction: 0x464c4f57

```
Before the proposed changes, and the following afterwards:
```
000007e4 <wolfBoot_get_blob_version>:
     7e4:   2300        movs    r3, #0
     7e6:   b507        push    {r0, r1, r2, lr}
     7e8:   6802        ldr r2, [r0, #0]
     7ea:   9301        str r3, [sp, #4]
     7ec:   4b09        ldr r3, [pc, #36]   ; (814 <wolfBoot_get_blob_version+0x30>)
     7ee:   429a        cmp r2, r3
     7f0:   d003        beq.n   7fa <wolfBoot_get_blob_version+0x16>
     7f2:   2000        movs    r0, #0
     7f4:   b003        add sp, #12
     7f6:   f85d fb04   ldr.w   pc, [sp], #4
     7fa:   2101        movs    r1, #1
     7fc:   aa01        add r2, sp, #4
     7fe:   3008        adds    r0, #8
     800:   f7ff ffc8   bl  794 <wolfBoot_find_header>
     804:   2800        cmp r0, #0
     806:   d0f4        beq.n   7f2 <wolfBoot_get_blob_version+0xe>
     808:   9b01        ldr r3, [sp, #4]
     80a:   2b00        cmp r3, #0
     80c:   d0f1        beq.n   7f2 <wolfBoot_get_blob_version+0xe>
     80e:   9b01        ldr r3, [sp, #4]
     810:   6818        ldr r0, [r3, #0]
     812:   e7ef        b.n 7f4 <wolfBoot_get_blob_version+0x10>
     814:   464c4f57            ; <UNDEFINED> instruction: 0x464c4f57
```

danielinux, 2022-11-28:
This should be fixed now via #251 which has been merged today. Feel free to reopen is anything does not check out.

Thanks!

-- 
D.

closed 2022-11-28 by danielinux