wolfSSL/wolfBoot #415: KeyGenTool Have Some Problem

lifeng5478, 2024-03-04:
key_import does not match RSA file import

and RSA2048 KeySize not 320.its 294

[image]
[image]
and then another thing,,,,uhmmm. maybe not a question。 
[image]
because i do not use the special section to set this keystore.

jpbland1, 2024-03-11:
Hey @lifeng5478,

Could you share what version of wolfBoot these screenshots are from? It doesn't match what I'm seeing in our keytools, and I know this has been refactored recently. When that key_import code gets called the public key is already loaded into memory and for RSA is already in the correct format to use so it will be a little different from our normal file import procedure.

I'm not sure what you're asking about the keystore, are you saying you don't use keyimport.c for your keystore and have configured something else?

Best Wishes,
John Bland

danielinux, 2024-03-22:
Hello @lifeng5478 

Did you have time to check @jpbland1 's comments above? Which version of wolfBoot are you running? How are you importing the key from the keystore structure?

danielinux, 2024-05-14:
Closing due to lack of detailed information. Feel free to reopen if you see the issue again.

closed 2024-05-14 by danielinux