wolfSSL/wolfBoot #426: Bootloader firmware not built without test-apps / cannot reconfigure cmake build

opkaizen, 2024-04-05:
Hi!
I have issues when trying to build the bootloader firmware via cmake.

I tried to build the project using the following command:

```
mkdir build && cd build && cmake -DWOLFBOOT_TARGET=stm32h7 -DWOLFBOOT_PARTITION_BOOT_ADDRESS=0x8020000 -DWOLFBOOT_SECTOR_SIZE=0x20000 -DWOLFBOOT_PARTITION_SIZE=0xD0000 -DWOLFBOOT_PARTITION_UPDATE_ADDRESS=0x80F0000 -DWOLFBOOT_PARTITION_SWAP_ADDRESS=0x81C0000 .. && make
```

It's the example command available in the README.md file without the -DTEST_APPS flag to avoid building the baremetal test application, as I only want the bootloader.bin. With this, `wolfboot.bin` and `wolfboot.elf` are not being built (but they are in the Makefile version: `cp config/examples/stm32h7.config .config && make wolfboot.bin`). How can I avoid building the test apps when I only want the wolfboot bootloader firmware using cmake?

The logs are:
```
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building for stm32h7
-- Cross-compiling using GNU arm-none-eabi toolchain
-- Signing image using ECC256
-- Using SHA256 hash
-- Using C Keytools
-- Using fast math
-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: /<REDACTED>/wolfBoot/build
[...]
[  5%] Building C object CMakeFiles/wolfboothal.dir/hal/stm32h7.c.o
[ 10%] Linking C static library libwolfboothal.a
[ 10%] Built target wolfboothal
[ 15%] Building signing tool
[ 21%] Building keygen tool
[ 21%] Built target keytools
[ 26%] Generating keystore.c and signing private key
<REDACTED>
[ 31%] Building C object CMakeFiles/public_key.dir/keystore.c.o
[ 36%] Linking C static library libpublic_key.a
[ 42%] Built target public_key
[ 47%] Building C object lib/CMakeFiles/wolfcrypt.dir/wolfssl/wolfcrypt/src/integer.c.o
[ 52%] Building C object lib/CMakeFiles/wolfcrypt.dir/wolfssl/wolfcrypt/src/tfm.c.o
[ 57%] Building C object lib/CMakeFiles/wolfcrypt.dir/wolfssl/wolfcrypt/src/ecc.c.o
[ 63%] Building C object lib/CMakeFiles/wolfcrypt.dir/wolfssl/wolfcrypt/src/memory.c.o
[ 68%] Building C object lib/CMakeFiles/wolfcrypt.dir/wolfssl/wolfcrypt/src/wc_port.c.o
[ 73%] Building C object lib/CMakeFiles/wolfcrypt.dir/wolfssl/wolfcrypt/src/wolfmath.c.o
[ 78%] Building C object lib/CMakeFiles/wolfcrypt.dir/wolfssl/wolfcrypt/src/hash.c.o
[ 84%] Building C object lib/CMakeFiles/wolfcrypt.dir/wolfssl/wolfcrypt/src/sha256.c.o
[ 89%] Linking C static library libwolfcrypt.a
[ 89%] Built target wolfcrypt
[ 94%] Building C object CMakeFiles/wolfboot.dir/src/libwolfboot.c.o
[100%] Linking C static library libwolfboot.a
[100%] Built target wolfboot
```

I also have issues when I try to reconfigure the project. Rerunning the cmake command above generates the following logs:
```
✦ ❯ cmake -DWOLFBOOT_TARGET=stm32h7 -DWOLFBOOT_PARTITION_BOOT_ADDRESS=0x8020000 -DWOLFBOOT_SECTOR_SIZE=0x20000 -DWOLFBOOT_PARTITION_SIZE=0xD0000 -DWOLFBOOT_PARTITION_UPDATE_ADDRESS=0x80F0000 -DWOLFBOOT_PARTITION_SWAP_ADDRESS=0x81C0000 ..
-- Building for stm32h7
-- Cross-compiling using GNU arm-none-eabi toolchain
-- Signing image using ECC256
-- Using SHA256 hash
-- Using C Keytools
-- Using fast math
-- Configuring done (0.0s)
You have changed variables that require your cache to be deleted.
Configure will be re-run and you may have to reset some variables.
The following variables have changed:
CMAKE_C_COMPILER= /usr/bin/arm-none-eabi-gcc
CMAKE_CXX_COMPILER= /usr/bin/arm-none-eabi-g++

-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - broken
CMake Error at /usr/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/usr/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/<REDACTED>/wolfBoot/build/CMakeFiles/CMakeScratch/TryCompile-e4htJH'
    
    Run Build Command(s): /usr/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile cmTC_3a1e1/fast
    /usr/bin/make  -f CMakeFiles/cmTC_3a1e1.dir/build.make CMakeFiles/cmTC_3a1e1.dir/build
    make[1]: Entering directory '/<REDACTED>/wolfBoot/build/CMakeFiles/CMakeScratch/TryCompile-e4htJH'
    Building C object CMakeFiles/cmTC_3a1e1.dir/testCCompiler.c.o
    /usr/bin/arm-none-eabi-gcc    -o CMakeFiles/cmTC_3a1e1.dir/testCCompiler.c.o -c /<REDACTED>/wolfBoot/build/CMakeFiles/CMakeScratch/TryCompile-e4htJH/testCCompiler.c
    Linking C executable cmTC_3a1e1
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_3a1e1.dir/link.txt --verbose=1
    /usr/bin/arm-none-eabi-gcc CMakeFiles/cmTC_3a1e1.dir/testCCompiler.c.o -o cmTC_3a1e1
    /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/libc.a(libc_a-exit.o): in function `exit':
    /build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/newlib/../../../newlib-4.3.0.20230120/newlib/libc/stdlib/exit.c:65:(.text.exit+0x28): undefined reference to `_exit'
    /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/libc.a(libc_a-closer.o): in function `_close_r':
    /build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/newlib/../../../newlib-4.3.0.20230120/newlib/libc/reent/closer.c:47:(.text._close_r+0x18): undefined reference to `_close'
    /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/libc.a(libc_a-lseekr.o): in function `_lseek_r':
    /build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/newlib/../../../newlib-4.3.0.20230120/newlib/libc/reent/lseekr.c:49:(.text._lseek_r+0x24): undefined reference to `_lseek'
    /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/libc.a(libc_a-readr.o): in function `_read_r':
    /build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/newlib/../../../newlib-4.3.0.20230120/newlib/libc/reent/readr.c:49:(.text._read_r+0x24): undefined reference to `_read'
    /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/libc.a(libc_a-writer.o): in function `_write_r':
    /build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/newlib/../../../newlib-4.3.0.20230120/newlib/libc/reent/writer.c:49:(.text._write_r+0x24): undefined reference to `_write'
    /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/libc.a(libc_a-sbrkr.o): in function `_sbrk_r':
    /build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/newlib/../../../newlib-4.3.0.20230120/newlib/libc/reent/sbrkr.c:51:(.text._sbrk_r+0x18): undefined reference to `_sbrk'
    collect2: error: ld returned 1 exit status
    make[1]: *** [CMakeFiles/cmTC_3a1e1.dir/build.make:99: cmTC_3a1e1] Error 1
    make[1]: Leaving directory '/<REDACTED>/wolfBoot/build/CMakeFiles/CMakeScratch/TryCompile-e4htJH'
    make: *** [Makefile:127: cmTC_3a1e1/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:43 (project)

-- Configuring incomplete, errors occurred!
```

I guess cmake doesn't like the multiarch build between the static libs (arm-none-eabi) and the keytools.

Thanks,

lealem47, 2024-04-05:
Hi @opkaizen,

Thank you for this report. I will look into this.

lealem47, 2024-04-22:
Hi @opkaizen,

Can you give https://github.com/wolfSSL/wolfBoot/pull/441 a try? You would just have to add `-DBUILD_IMAGE=yes` to your command

lealem47, 2024-04-22:
About the failure on re-running cmake, it's an issue with the cache from the first run. If you delete `build/CMakeCache.txt`, you should be able to re-run the cmake command and see it succeed

danielinux, 2024-05-14:
I'm closing this in favor of #441 (if I understand correctly this was initially handled in #427). Feel free to continue the discussion there.

closed 2024-05-14 by danielinux