wolfSSL/wolfBoot #438: i.MX RT10xx operator '*' has no right operand

MulattoKid, 2024-04-19:
When compiling for i.MX RT10xx I get the following error
```
core/devices/MIMXRT1042/MIMXRT1042.h:32067:65: error: operator '*' has no right operand
32067 | #define LPUART1                                  ((LPUART_Type *)LPUART1_BASE)
```

I've not hit this problem before, as we've been running with a slightly custom wolfBoot version internally.

dgarske, 2024-04-19:
@bigbrett is going to review this. 

Does this patch resolve the error? I can't tell what the specific wolfBoot error is, but I suspect it's this macro check.

```
diff --git a/hal/imx_rt.c b/hal/imx_rt.c
index 3c6e7398..458151b7 100644
--- a/hal/imx_rt.c
+++ b/hal/imx_rt.c
@@ -715,7 +715,7 @@ void uart_init(void)
     lpuart_config_t config;
     uint32_t uartClkSrcFreq = 20000000U; /* 20 MHz */
 
-#if UART_BASEADDR == LPUART1
+#if UART_BASEADDR == LPUART1_BASE
     /* Configure the UART IO pins for LPUART1
      * Tested with RT1040, RT1050, RT1062 and RT1064
      */
```

Thanks,
David Garske, wolfSSL

MulattoKid, 2024-04-19:
Yes, it's `UART_BASEADDR == LPUART1` that caused the issue.

dgarske, 2024-04-19:
Fixed with PR #439.

closed 2024-04-19 by dgarske