wolfSSL/wolfBoot #448: RZN2L - Separating App and Loader with FreeRTOS

KeuntekLee, 2024-05-07:
Hi.

I have some question about RZN2L example codes.

Examples in this repository for RZN2L, the app is created with Baremetal project.

Is this example and features (Separating App and Loader) can be done with App program with FreeRTOS?

BR

dgarske, 2024-05-07:
Hi BR (@KeuntekLee ),

Will you confirm you are using the latest RZ work in https://github.com/wolfSSL/wolfBoot/pull/444 ?

wolfBoot itself is RTOS agnostic, so using FreeRTOS with your own application is supported.

@miyazakh will you help support this ticket?

Thanks,
David Garske, wolfSSL

miyazakh, 2024-05-08:
Hi BR(@KeuntekLee),

 Thank you for your interest in our wolfBoot example on Renesas RZ. As David mentions, wolfBoot is independent to RTOS. I think that the app can be run under RTOS though startup code might need to be tweaked. In addition to that, in this example, the app is downloaded into System RAM, which has enough memory space to run it with RTOS.

KeuntekLee, 2024-05-09:
Hi BR(@miyazakh ),

Thanks for the comments. In my case, the app with FreeRTOS is less than 200Kb, so I think it might fit in the System RAM area.
As you mentioned, the startup code needs to be modified when the app is not the baremetal-based. I fully understand that wolfBoot is not dependent on this scenario, and just ask about some examples on this case for reference (FreeRTOS app + xspi bootloader). If you have future item of this, please let me know.

miyazakh, 2024-05-09:
Hi BR(@KeuntekLee)

 Unfortunately, I don't have an example that runs it with RTOS now. The easiest way for a trial is to use [the example](https://github.com/wolfSSL/wolfBoot/tree/master/IDE/Renesas/e2studio/RZN2L) in wolfBoot rep and download your example instead of the original app.

closed 2024-12-31 by danielinux