wolfSSL/wolfBoot #452: Log to flash support

MulattoKid, 2024-05-30:
Hi,

This isn't an issue, but rather a question regarding adding support for logging output from `wolfBoot_printf` to flash.

The use is twofold (particularly with units in the field in mind):
- if the bootloader fails to boot any partition, we could manually scrape the flash region used for logging to see what happened
- if an application update is performed, and the new application version is either rejected by the bootloader or doesn't confirm the update, a rollback is performed. When the old application version boots again, it could scrape the flash region used for logging to see what happened

My suggestion would be to add something like the following options to the configuration file:
- `WOLFBOOT_LOG_FILE_ENABLED`
- `WOLFBOOT_LOG_FILE_ADDRESS` (needs to be aligned with `WOLFBOOT_SECTOR_SIZE`)
- `WOLFBOOT_LOG_FILE_SIZE` (needs to be a multiple of `WOLFBOOT_SECTOR_SIZE`)

I have the following questions:
1. Is this functionality something you'd consider accepting into wolfBoot?
2. If yes to 1., would you be able to add support?

cc @danielinux :slightly_smiling_face: 

Kind regards,
Daniel

danielinux, 2024-06-26:
Hi @MulattoKid 

I've added this feature request to our list. Thank you!

closed 2024-06-26 by danielinux

MulattoKid, 2024-06-27:
Thanks @danielinux! Any idea on timeline for support?

MulattoKid, 2025-02-24:
Hi, just wondering if there's any ETA for this feature, or is it quite low on the list?