wolfSSL/wolfBoot #453: Boot flags update fail on H7 (if last sector has some data)

ClarkyKent, 2024-06-03:
**### Description**

This issue happens in STM32H72x/3x microcontroller where it has only 1 MB of flash (hence only one bank) and Sector is 128K in size. 

When code size (application) is 260 KBs or more, the memory map in linker looks like this.

| Bootloader (Sector 1)     |
|---------------------------|
| Application (sector 2-4)  |
| Update area (sector 5-7) |
| Swap area (Sector 8)      |

So, when update FW is more than 256 KBs in size it will write last few bytes in 3rd sector (Sector 7 in case above). Which is the same sector, which has end address for FLAGS. 
So, when system reboots after DFU and our update FW lies in `Update Area` it will erase the last sector to update FLAGS. Which is a problem since it will erase FW data which is in start of sector 7.  (see HAL implementation of [H7 here](https://github.com/wolfSSL/wolfBoot/blob/53a7f643b28f8738b31357d13a051bab79c28bf6/hal/stm32h7.c#L124)

### **Possible solution**

We can use SWAP area as temp buffer where we write last few bytes of FW in sector 7. And then erase last sector, then update FLAGS (in cache) and write everything back to last sector.

dgarske, 2024-06-03:
Hi @ClarkyKent ,

Thank you for this report. The sector size is a common issue with the H7.  Do you need power fail safety on your device? If not you could use `DISABLE_BACKUP=1`.

We have discussed internally some ideas for how to improve this use-case and I expect we'll have a better solution available soon.

@danielinux do you have any other ideas to solve?

Thanks,
David Garske, wolfSSL

danielinux, 2024-06-18:
The default to guarantee power-fail updates in such a configuration requires two sectors. 

This is particularly unfortunate on the H7, which has both a FLASH memory not allowing multiple write after erase (and for this reason, requiring NVM_FLASH_WRITEONCE), and at the same time a huge sector size (128KB), making it uncomfortable for wolfBoot to acommodate the fail-safe/backup feature.

Using NO_BACKUP, the update is directly copied into the boot partition, removing the requirement for the extra sectors (and for the SWAP partition, entirely). An alternative might be using an external flash for UPDATE/SWAP.

closed 2024-06-26 by danielinux