wolfSSL/wolfBoot #471: Feature request: SEGGER RTT logging

vhdirk, 2024-07-18:
At @Aloxy-io, we're using SEGGER RTT for logging. We've been experimenting with using that for wolfboot, too. Is this something that could be considered for inclusion? We have some proof of concept implementation that we're willing to contribute, but the dev team isn't quite sure if the licenses are compatible (we're not lawyers).

danielinux, 2024-07-18:
Hi @vhdirk,
Thanks for sharing this with us. I think it might be possible to integrate a custom wolfBoot_printf that redirects the output to the RTT channel. I'm not aware of the license of the RTT code, but adding the interface might be OK as long as we don't directly distribute third party code.

danielinux, 2024-07-18:
The team is suggesting to share the code privately first via support@ if you have any doubts about the license, so we can check any issues before publishing code or considering for inclusion.
Thanks!

danielinux, 2024-08-22:
It's clear how to proceed with this now. The integration of RTT logging will be resumed later. Closing for now. (zd18370)

closed 2024-08-22 by danielinux