wolfSSL/wolfBoot #489: Non-RAM functions in hal_flash_erase/write on iMX RT1060

MulattoKid, 2024-08-15:
Hi,

After increasing the value of `WOLFBOOT_SECTOR_SIZE` to reduce the wear on the flash sectors used for swap during updates, we've started to experience issues with hangs in `hal_flash_erase` and `hal_flash_write`. This happens occasionally when starting from a totally erased flash, and then programming the bootloader and application image. It happens when we call `wolfBoot_success`, as it causes erases and writes to the last two sectors in the BOOT and UPDATE partitions (wolfBoot housekeeping).

Upon investigation, we've narrowed it down to the use of `memcpy` and `DCACHE_InvalidateByRange` in both flash functions. I've previously added disabling of interrupts and invalidating the DCache in these functions to ensure no access over FlexSPI while erasing/writing to flash and data coherency. However, the functions used here should also be in RAM, so as to not perform accesses over FlexSPI.

**memcpy**
wolfBoot has its own `memcpy` implementation which is declared with `RAMFUNCTION`. So long as both `__WOLFBOOT` and `RAM_CODE` are defined when compiling wolfBoot, this is fine. However, we define `WOLFBOOT_USE_STDLIBC`, which means our newlib `memcpy` is used, which gets placed in flash, not RAM. This is also true when compiling libwolfBoot for our application. We therefore had to add the following to wolfBoot's linker script:
```
    .text :
    {
      <..>
      *(EXCLUDE_FILE(
          *libc.a:lib_a-memcpy.o
      ) .text)                 /* .text sections (code) */
      *(EXCLUDE_FILE(
          *libc.a:lib_a-memcpy.o
      ) .text*)                /* .text* sections (code) */
      *(EXCLUDE_FILE(
          *libc.a:lib_a-memcpy.o
      ) .rodata)               /* .rodata sections (constants, strings, etc.) */
      *(EXCLUDE_FILE(
          *libc.a:lib_a-memcpy.o
      ) .rodata*)              /* .rodata* sections (constants, strings, etc.) */
      <...>
    } > FLASH

    .data : AT (_stored_data)
    {
      <...>
      KEEP(*libc.a:lib_a-memcpy.o (.text*))
      <...>
    } > RAM
```

and this to our link script to ensure this function is placed in RAM instead of flash:
```
    .text :
    {
      <...>
      *(EXCLUDE_FILE(
          *libc_nano.a:lib_a-memcpy.o
      ) .text)                 /* .text sections (code) */
      *(EXCLUDE_FILE(
          *libc_nano.a:lib_a-memcpy.o
      ) .text*)                /* .text* sections (code) */
      *(EXCLUDE_FILE(
          *libc_nano.a:lib_a-memcpy.o
      ) .rodata)               /* .rodata sections (constants, strings, etc.) */
      *(EXCLUDE_FILE(
          *libc_nano.a:lib_a-memcpy.o
      ) .rodata*)
      <...>
    } > m_text

    .data : AT(__DATA_ROM)
    {
      <...>
      *libc_nano.a:lib_a-memcpy.o (.text*)
      <...>
    } > m_dtcm
```

**DCACHE_InvalidateByRange**
The same goes for `DCACHE_InvalidateByRange`, except that there's no handling of that currently. We've based our solution on NXP's linker scripts where the entire compiled object for `fsl_cache.c` is omitted from being placed in flash, and is instead put in RAM:
```
      <...>
      *(EXCLUDE_FILE(
          */fsl_cache.c.obj
      ) .text)                 /* .text sections (code) */
      *(EXCLUDE_FILE(
          */fsl_cache.c.obj
      ) .text*)                /* .text* sections (code) */
      *(EXCLUDE_FILE(
          */fsl_cache.c.obj
      ) .rodata)               /* .rodata sections (constants, strings, etc.) */
      *(EXCLUDE_FILE(
          */fsl_cache.c.obj
      ) .rodata*)              /* .rodata* sections (constants, strings, etc.) */
      <...>

      <...>
      KEEP(*/fsl_cache.o(.text .text* .rodata .rodata*))
      <...>
```

A colleague of mine has made a [fix](https://github.com/aminasander/wolfboot-ram/commit/63f09f37d01027fdc6f40613858110f47d450db4) that should solve the issue.

Does this issue make sense to you, and does the solution seem acceptable? If so, I can make a PR to get this in :slightly_smiling_face: 

Thanks,
Daniel

bigbrett, 2024-08-15:
@MulattoKid is there a particular reason you don't want wolfBoot's built-in string functions? Just curious what the restriction is.

MulattoKid, 2024-08-16:
> @MulattoKid is there a particular reason you don't want wolfBoot's built-in string functions? Just curious what the restriction is.

I remember adding `WOLFBOOT_USE_STDLIBC` as part of our attempt of speeding up the boot process. I don't have data about the change in speed for that particular change.

MulattoKid, 2024-08-19:
After doing the things mentioned above we still ran into the same issue. I believe there's a bug in my previous PRs to add support for other RT targets, where the `USE_GET_CONFIG` macro isn't defined for all RT targets. When this isn't defined the flash config in flash is used, which also violates the rule about not accessing flash when erasing/programming. Does this also sound correct?

bigbrett, 2024-08-19:
Hi @MulattoKid,

Sorry, I must be missing something here - have you defined `USE_GET_CONFIG` for those targets, and can confirm the problem has gone away? Or do you think that should fix it, but when testing it doesn't, and you are asking why? 

I do agree in principle that executing code from flash that does programming and erases needs to be avoided while XIP.

MulattoKid, 2024-08-20:
Sorry, I'll try to clarify.

My initial points regarding functions needing to be in RAM while erasing/programming flash when using XIP still stands. I think both `memcpy` and `DCACHE_InvalidateByRange` are candidates for this as they're inside the disabling and enabling of interrupts that happens before program/erase are called, i.e.:
``` C
/**
 * Disable interrupts before accessing flash when using XIP
 * (note 4 p.279 in i.MX RT1060 Processor Reference Manual, Rev. 3, 07/2021)
 */
asm volatile("cpsid i");
status = g_bootloaderTree->flexSpiNorDriver->erase(0, FLEXSPI_CONFIG, address - FLASH_BASE, len);
/**
 * Flash is memory mapped, so the address range must be invalidated in data cache
 * to ensure coherency between flash and cache
 */
DCACHE_InvalidateByRange(address, len);
asm volatile("cpsie i");
```
By default wolfBoot handles `memcpy`, but as we define `WOLFBOOT_USE_STDLIBC` for better performance we had to put it in RAM our selves. `DCACHE_InvalidateByRange` is not in RAM by default in wolfBoot.

Regarding `USE_GET_CONFIG`: the same logic applies to data that can be accessed during erase/programming of flash. I believe that the `FLEXSPI_CONFIG` that's passed to the erase/program calls can also trigger this issue. For most `CPU_MIMXRT10**` targets, `USE_GET_CONFIG` is defined. However, at least for the two targets I added (`CPU_MIMXRT1042XJM5B` and `CPU_MIMXRT1061CVJ5B`), that's not the case, i.e.:
```C
#ifdef CPU_MIMXRT1064DVL6A
#include "evkmimxrt1064_flexspi_nor_config.h"
#define USE_GET_CONFIG
#endif
#ifdef CPU_MIMXRT1062DVL6A
#include "evkmimxrt1060_flexspi_nor_config.h"
#define USE_GET_CONFIG
#endif
#ifdef CPU_MIMXRT1062DVL6B
#include "evkbmimxrt1060_flexspi_nor_config.h"
#define USE_GET_CONFIG
#endif
#ifdef CPU_MIMXRT1061CVJ5B
#include "evkmimxrt1060_flexspi_nor_config.h"
#endif
#ifdef CPU_MIMXRT1052DVJ6B
#include "evkbimxrt1050_flexspi_nor_config.h"
#endif
#ifdef CPU_MIMXRT1042XJM5B
#include "evkmimxrt1040_flexspi_nor_config.h"
#endif
```
By defining `USE_GET_CONFIG` for `CPU_MIMXRT1061CVJ5B` the crash we've seen seems to go away.

To summarize, I believe that `USE_GET_CONFIG` should be added to all RT targets, and that `memcpy` (also when `WOLFBOOT_USE_STDLIBC` is defined) and `DCACHE_InvalidateByRange` need to be placed in RAM. For our application we can just modify what we need internally, but there are some changes required in the linker script for wolfBoot.

Hopefully that's clarifying :slightly_smiling_face:

bigbrett, 2024-08-20:
Yes, it is, thanks! I mostly just wanted to make sure you had found an appropriate fix for your specific case, before integrating a general solution. 

I will look at making a PR that adds `USE_GET_CONFIG` to all RT targets, and adds the linker file modifications to place both the libc memcpy and DCACHE functions into the RAMCODE section.

Out of curiosity, how frequently are you seeing the crash?

MulattoKid, 2024-08-20:
Oh, a colleague of mine just said he ran into the same issue again, even with both of the fixes mentioned above. That really surprises me...do you any other ideas of what could be causing this issue? Just to reiterate, this started happening when increasing `WOLFBOOT_SECTOR_SIZE` beyond the flash sector size.

bigbrett, 2024-08-20:
That is troubling.... Hmm no ideas off the top of my head, but I will be starting to dig into this soon. 

Could you share your .config and any other non-standard defines you are setting? Also please share the MCUXpresso SDK version so we can try and replicate with the exact same SDK package. 

And this has so far only been replicated on `CPU_MIMXRT1042XJM5B` and `CPU_MIMXRT1061CVJ5B`?

MulattoKid, 2024-08-21:
Thanks for following this up, we'll continue looking as well. To your questions:
- Config:
```
# Platform
ARCH?=ARM
TARGET?=imx_rt
MCUXPRESSO_CPU?=MIMXRT1061CVJ5B
MCUXSDK?=1
MCUXPRESSO?=$(SdkRootDirPath)/core
MCUXPRESSO_DRIVERS?=$(MCUXPRESSO)/devices/MIMXRT1061
MCUXPRESSO_CMSIS?=$(MCUXPRESSO)/CMSIS
VTOR?=1
CORTEX_M0?=0
NO_ASM?=0
NO_MPU=1
SPI_FLASH?=0
NVM_FLASH_WRITEONCE?=1
V?=0
SPMATH?=1
RAM_CODE?=0
DUALBANK_SWAP?=0
PKA?=1

# wolfBoot
WOLFBOOT_VERSION?=100663296

# Flash
WOLFBOOT_PARTITION_SIZE?=0x300000
WOLFBOOT_SECTOR_SIZE?=0x10000
WOLFBOOT_PARTITION_BOOT_ADDRESS?=0x60010000
WOLFBOOT_PARTITION_UPDATE_ADDRESS?=0x60310000
WOLFBOOT_PARTITION_SWAP_ADDRESS?=0x60610000
WOLFBOOT_SMALL_STACK?=1
# Flash Options
CFLAGS_EXTRA+=-DCONFIG_FLASH_W25Q128JV -DUART_BASE=LPUART4_BASE -DUART_BASEADDR=LPUART4 -DWOLFBOOT_USE_STDLIBC -DWOLFBOOT_SHA_BLOCK_SIZE=4096 -DDCP_USE_DCACHE=0

# Updates
ALLOW_DOWNGRADE?=1
DELTA_UPDATES?=1

# Debug
DEBUG?=1
DEBUG_SYMBOLS?=1
DEBUG_UART?=1
PRINTF_ENABLED?=1

# Signature
SIGN?=ED25519
HASH?=SHA256

# Header
IMAGE_HEADER_SIZE?=1024
```
- NXP SDK v2.15
- We've only observed it on `CPU_MIMXRT1061CVJ5B`
- This is a custom PCB, and we're using `Winbond W25Q128JV` (`CONFIG_FLASH_W25Q128JV`)
- No exception seemed to be triggered, and I believe it got stuck int he FlexSPI ROM API

I did some more testing this morning, and here's what I've found:
1. If I take latest of wolfBoot and add `USE_GET_CONFIG` and use our latest `main` (no changes related to `memcpy` and `DCACHE_InvalidateByRange`), things seem to run fine - I can't reproduce what my colleague experienced 
2. My colleague has made a copy of wolfBoot, and made the changes mentioned regarding `memcpy` and `DCACHE_InvalidateByRange`. He also has a branch of our application with similar changes. When running with those versions + adding `USE_GET_CONFIG` I consistently get stuck in `hal_flash_init`, so there seems to be something wrong with our changes here.

MulattoKid, 2024-08-21:
### Scenario 1 ###
- wolfBoot: `40ccd9ea220e90fb567321a6c8e6c33336d12688`
- `main` of our application
- `.config` as above
- Erase flash using LinkServer
- Flash bootloader without letting it run
- Flash app without letting it run
- Power cycle MCU -> it sometimes gets stuck in erase/program (it varies)
  - Power cycling the MCU again and things run fine (this is consistently true)

### Scenario 2 ###
- wolfBoot: `40ccd9ea220e90fb567321a6c8e6c33336d12688` with added `USE_GET_CONFIG` to remaining RT targets
- `main` of our application
- `.config` as above
- Erase flash using LinkServer
- Flash bootloader without letting it run
- Flash app without letting it run
- Power cycle MCU -> works fine

I've tested 10 times, and not hit the issue. Normally hitting the issue only takes 1-3 attempts. This doesn't necessarily mean it completely fixes the issue, but it might make it happen a lot less often at least. The changes wrt. `memcpy` and `DCACHE_InvalidateByRange` might still be needed?

### Scenario 3 ###
- wolfBoot: `40ccd9ea220e90fb567321a6c8e6c33336d12688` with added `USE_GET_CONFIG` to remaining RT targets
- `fix_memcpy_and_dcache` branch of our application
  - Puts `memcpy` and `DCACHE_InvalidateByRange` in RAM for our app (and libwolfBoot), but not wolfBoot
- `.config` as above
- Erase flash using LinkServer
- Flash bootloader without letting it run
- Flash app without letting it run
- Power cycle MCU -> gets stuck in `hal_flash_init`
  - Power cycling the MCU again results in the same thing

### Summary ###
To me it seems that adding `USE_GET_CONFIG` is needed (or is at least a way of solving parts of the problem related to accessing flash at the same time as it's being erased/programmed).

I will look at why the changes to our application in Scenario 3 results in issues in `hal_flash_init`.

bigbrett, 2024-08-21:
Thank you so much for that information, it helps a lot. 

> Finally, I do believe a similar change is needed in hal_flash_init as was added to hal_flash_erase/write, with disabling/enabling interrupts before/after calling the FlexSPI ROM API. Otherwise, accesses to flash could happen at the same time.
Yes, I concur on this. 

I'll let you know what we find as we dig deeper into this. Thanks again.

bigbrett, 2024-08-21:
I see you don't have `RAM_CODE=1` defined in your config. Is that set elsewhere? That should be set by default for all the imx targets, given that we realized these issues are likely caused by XIP flash code touching flash

> I do wonder why the flash configuration for the RT1052* was different, and why it wasn't based on the same strategy. I haven't looked in the reference manual for the 1052, but they should be pretty similar. A suggestion I have is to change at least the flash configuration for the RT1061* to the same as the other RT106x* targets, and rely on the JEDEC commands to retrieve the rest. We've tested this, and see no difference in behavior. This also requires adding the USE_GET_CONFIG to that target. This could possibly be done for the RT1040/1050 targets as well, but I'm not 100% sure.

I agree, and Interestingly, when I add `USE_GET_CONFIG` to the 1052 target, an exception is triggered in `hal_flash_init()` when it calls the ROM `get_config()` API.

I should note that I am seeing similar weirdness in the stock imx RT builds, so I'm not sure the std libc have anything to do with it anymore - though I agree that those should be in RAM regardless.

bigbrett, 2024-08-21:
> I agree, and Interestingly, when I add USE_GET_CONFIG to the 1052 target, an exception is triggered in hal_flash_init() when it calls the ROM get_config() API.

After a bit of digging, it seems the 1050 ROM API does not support `USE_GET_CONFIG`. So that is why

MulattoKid, 2024-08-22:
Yes, seems your are correct about the 1050. From their reference manuals I see this:

**1050**:
```
typedef struct
{
    uint32_t version;
    status_t (*init)(uint32_t instance, flexspi_nor_config_t *config);
    status_t (*program)(uint32_t instance, flexspi_nor_config_t *config, uint32_t dst_addr, const uint32_t *src);
    status_t (*erase_all)(uint32_t instance, flexspi_nor_config_t *config);
    status_t (*erase)(uint32_t instance, flexspi_nor_config_t *config, uint32_t start, uint32_t lengthInBytes);
    status_t (*read)(uint32_t instance, flexspi_nor_config_t *config, uint32_t *dst, uint32_t addr, uint32_t lengthInBytes);
    void (*clear_cache)(uint32_t instance);
    status_t (*xfer)(uint32_t instance, flexspi_xfer_t *xfer);
    status_t (*update_lut)(uint32_t instance, uint32_t seqIndex, const uint32_t *lutBase, uint32_t seqNumber);
} flexspi_nor_driver_interface_t;
```
**1040/1060**:
```
typedef struct
{
    uint32_t version;
    status_t (*init)(uint32_t instance, flexspi_nor_config_t *config);
    status_t (*program)(uint32_t instance, flexspi_nor_config_t *config, dst_addr, const uint32_t *src);
    status_t (*erase_all)(uint32_t instance, flexspi_nor_config_t *config);
    status_t (*erase)(uint32_t instance, flexspi_nor_config_t *config, uint32_t uint32_t lengthInBytes);
    status_t (*read)(uint32_t instance, flexspi_nor_config_t *config, uint32_t *dst, uint32_t addr, uint32_t lengthInBytes);
    void (*clear_cache)(uint32_t instance);
    status_t (*xfer)(uint32_t instance, flexspi_xfer_t *xfer);
    status_t (*update_lut)(uint32_t instance, uint32_t seqIndex, const uint32_t *lutBase, uint32_t seqNumber);
    status_t (*get_config)(uint32_t instance, flexspi_nor_config_t *config,
    serial_nor_config_option_t *option);
} flexspi_nor_driver_interface_t;
```

So the config for 1050 needs to stay, whereas for 1040 and 106x the `get_config` API call can be used.

bigbrett, 2024-08-22:
@MulattoKid thanks for those PRs! Just to confirm, do those PRs fix the hanging you are seeing in the ROM APIs on the 1060? Or is that still an issue and these are just intermediate fixes?

MulattoKid, 2024-08-22:
Thinking about it, I wonder if disabling interrupts isn't enough, do we need a full barrier as well to ensure correct behavior? `hal_flash_init` is missing both things, and barriers are missing before calling `DCache_InvalidateRange`. Also, `hal_flash_init` should probably be marked with `RAMFUNCTION`?

bigbrett, 2024-08-22:
@MulattoKid I made the exact same comment on #493 without having read this. Same wavelength! I agree that a memory barrier is likely needed and could be the source of the problem

MulattoKid, 2024-08-23:
I moved the two old PRs into a single one, with the addition of the DSB barrier: https://github.com/wolfSSL/wolfBoot/pull/495
Have tested, and have not ran into the issue reported here.

bigbrett, 2024-08-26:
Merged #495, issue reported fixed by adding appropriate fencing/barrier instructions after flash ROM API calls.

@MulattoKid thanks for the help in investigating this. Were you able to test this on the 1040 family too?

MulattoKid, 2024-08-27:
No, I've only tested it on the 1061 (our custom PCB).

No worries, thanks for being very quick to respond to all our questions and PRs :slightly_smiling_face:

closed 2024-08-27 by MulattoKid