wolfSSL/wolfBoot #506: wolfBoot header location

MulattoKid, 2024-09-20:
Hi,

I'm working on the i.MX RT10xx platform, and wanting to use NXP's Secure Provisioning Tool (SPT) for production. I've previously made a PR to wolfBoot to compile it for the IMX platform without the flash config and IVT+BootData+DCD, so that the SPT would accept the wolfBoot binary. I'm looking to also use on-the-fly decryption that the MCU supports, it'd be convenient if the tool also accepted our application binary as input, so that it could encrypt it and write it to flash.

However, the SPT doesn't accept the binary, as the start of the binary contains the wolfBoot header and not interrupt vector table (it's looking for the address of the ResetISR). One solution to this would be to not have the wolfBoot header at the start of the application binary, but rather at the end. Is this something that's possible for you to add as a configuration option?

I'm a bit hesitant to start looking at changing such fundamentals of wolfBoot, so it'd be great if this is something that could be added :slightly_smiling_face:

Thanks,
Daniel

danielinux, 2024-09-20:
Hello Daniel,
The header-first in wolfBoot images is a design choice, and it would be an invasive change for the current codebase to move it elsewhere, although technically possible.
I would suggest you try and find an alternative for your use case, e.g. encrypt-then-sign, so you could verify the (unencrypted) header and load the encrypted image at the beginning of the aligned space for in-place decryption.

If this is not an option, we could still file a feature request. If this works for you, let us know through support@.

Thanks,

Daniele

closed 2024-09-20 by danielinux

MulattoKid, 2024-09-23:
Thanks for the quick response! Will do.