wolfSSL/wolfBoot #537: Variables in options.mk not set when included in tool Makefiles

MulattoKid, 2025-01-13:
Hi,

We've seen an issue where one of our developers is attempting to build wolfBoot in a Docker container built for a `aarch64` host. When doing so the build fails (specifically for the `bin-assemble` tool) with the stabs debug format not being supported. We did some digging, and found that it's due to this check in `options.mk`:
``` Makefile
# allow elf inclusion of debug symbols even with optimizations enabled
# make DEBUG_SYMBOLS=1
ifeq ($(DEBUG_SYMBOLS),1)
  CFLAGS+=-g
  ifeq ($(USE_GCC),1)
    CFLAGS+=-ggdb3
  else
    CFLAGS+=-gstabs
  endif
endif
```
When compiling the actual bootloader this works fine as this line is at the top of the `Makefile`: `USE_GCC?=1`. However, when compiling the tools the `options.mk` file is included without `USE_GCC` being set to `1`.

Is there a reason the stabs format is used specifically, or could it be replaced by `-g`, to use the default format on the host platform? Or should `USE_GCC` be set to `1`, because the `Makefile` [explicitly uses GCC](https://github.com/wolfSSL/wolfBoot/blob/5fc2a5057e408e262f8edfe0480988b746699fc4/tools/bin-assemble/Makefile#L5).
Just to note that this works fine in Docker containers built for a `amd64` host.

danielinux, 2025-02-05:
Hi @MulattoKid 
Would explicitly adding USE_GCC=1 in bin-assemble/Makefile fix the issue?

```diff
diff --git a/tools/bin-assemble/Makefile b/tools/bin-assemble/Makefile
index 2fda4157..ed78a96e 100644
--- a/tools/bin-assemble/Makefile
+++ b/tools/bin-assemble/Makefile
@@ -3,6 +3,7 @@
 -include ../../options.mk
 
 CC=gcc
+USE_GCC=1
 CFLAGS+=-Wall -g -ggdb
 EXE=bin-assemble
```

MulattoKid, 2025-02-06:
I think so, i.e. ignoring whether `USE_GCC` is already set, and forcing it to 1.

MulattoKid, 2025-02-11:
Hi,

My colleague isn't able to reproduce the original issue anymore - maybe his setup has changed.
However, I believe my previous statement was incorrect. `options.mk` uses the value of `USE_GCC`, so `USE_GCC` should be set before including `options.mk`.

As we're not seeing the issue anymore I'll leave it up to you whether you want to actually move forward with this, or just close it 🙂 

Thanks

dgarske, 2025-02-17:
Hi @MulattoKid , thank you. We will go ahead and close this ticket. Thanks, David Garske, wolfSSL

closed 2025-02-17 by dgarske