wolfSSL/wolfBoot #542: Following Tricore Guide now fails to build wolfboot

kuolemaaa, 2025-01-28:
Some days ago I built succesfully the wolfboot located at IDE/AURIX following the readme.

Today I wanted to replicate the same on another machine and when I try to build wolfBoot it gives me an error about missing `hardware/flash.h`

```
fatal error: hardware/flash.h: No such file or directory
make: *** [makefile:63: all] Error 2
make[1]: *** [wolfBoot/hal/subdir.mk:50: wolfBoot/hal/rp2350.o] Error 1
make[1]: *** Waiting for unfinished jobs....
Symbol 'XIP_BASE' could not be resolved
Symbol 'XIP_BASE' could not be resolved
```

Is it related to your latest pushes?

kuolemaaa, 2025-01-28:
I confirm. You fucked up someway somewhere from commit 456a6f8

bigbrett, 2025-01-29:
Hi @kuolemaaa 

Thanks for you interest in wolfBoot on AURIX. 

This issue is an unfortunate consequence of the AURIX development studio. The IDE will try to compile all files present unless explicitly excluded, but there is no way to exclude a directory while still allowing select files in that directory to be compiled. Therefore every time we add files to wolfBoot, it has the potential to break the AURIX build. 

We are working on a better long-term solution for this, but in the meantime, please just manually exclude the problematic source files from  your build by right clicking the files (`wolfBoot/hal/rp2350.c` and `wolfBoot/hal/nxp_ls1028a.c`) and selecting `Resource Configuration > exclude from build`. Make sure you exclude from both the Debug and Release GCC builds. This should fix the proble.

I'll make sure this gets fixed in the project settings soon, but this is an immediate workaround.

Best,
Brett

closed 2025-01-29 by dgarske