wolfSSL/wolfBoot #547: --sha-only with Ed25519 keystore fails

MulattoKid, 2025-01-31:
Hi,

There seems to be an issue generating the SHA digest when using the `--sha-only` argument to the `sign` tool with the Ed25519 algorithm. To replicate do the following:
``` bash
make keytools
tools/keytools/keygen --ed25519 -g test.der
tools/keytools/sign --ed25519 --sha256 --sha-only <input image> keystore.der 1
```

The output looks something like this:
``` bash
Update type:          Firmware
Input image:          <input image>
Selected cipher:      ED25519
Selected hash  :      SHA256
Private key:           ~/wolfBoot/keystore.der
Output digest:        <digest>.bin
Target partition id : 1 
Manifest header size: 1024
Found ED25519 key
Key decode error -140
```

From a little debugging it seems to fail at https://github.com/wolfSSL/wolfBoot/blob/519e3b721a6942a12324762a95110a3dd082f7e6/tools/keytools/sign.c#L580

Doing the same with but with `--ecc256` works. I don't think I'm doing anything different between the two.

Thanks

dgarske, 2025-01-31:
Hi @MulattoKid ,

It is failing because it expects the input key to be a public key when using `--sha-only`.

Please try the following:

```
tools/keytools/keygen --ed25519 -g test.der --exportpubkey
tools/keytools/sign --ed25519 --sha256 --sha-only <input image> test_pub.der 1
```

That worked for me.

Thanks,
David Garske, wolfSSL

MulattoKid, 2025-01-31:
Ah, I hadn't noticed that option before. However, I'm a bit confused as to why the way I did it works with ECC256 but not Ed25519?

To clarify, the `.der` file I passed to `sign` wasn't the output file I specified to `keygen`'s `-g` option. When I run `keygen -g test.der` I get three files: `test.der` (which is the private key), `keystore.der` and `keystore.c`. I thought that `keystore.der` would contain the public key? Have I misunderstood something?

dgarske, 2025-01-31:
Hi @MulattoKid ,

It's because our wolfCrypt ECC public key import code allows importing a public key when passing a private key. ED25519 API's do not.

Thanks,
David Garske, wolfSSL

MulattoKid, 2025-01-31:
Thanks for clarifying :)

closed 2025-01-31 by MulattoKid