wolfSSL/wolfBoot #554: Building and running the wolfBoot demo - Build Error

Jpgig, 2025-02-25:
As per the build instructions as below:
Build wolfBoot keytools and generate keys
1) Open a WSL2 terminal and navigate to the top level wolfBoot directory
2) Compile the keytools by running make keytools
When I performed second step getting below build error:

[image]

dgarske, 2025-02-25:
Hi @Jpgig ,

Please make sure and init and update the submodules.

```sh
git submodule init
git submodule update
```

The key tools use wolfSSL/wolfCrypt which are located in the lib/wolfssl submodule.

Thanks, David Garske, wolfSSL

Jpgig, 2025-02-25:
Thanks David, this resolved the issue.

closed 2025-02-25 by Jpgig