wolfSSL/wolfBoot #556: WolfBoot on Infineon AURIX TC3xx - Build error with GCC

Jpgig, 2025-02-26:
I'm trying to build the WolfBoot on Infineon AURIX TC3xx with GCC compiler but getting build errors as below:

GCC build error:

[image]

Jpgig, 2025-02-26:
I copied the entire wolfBoot folder from the WSL to the windows and then tried to build it. 
Seems its went through above errors but now getting few other errors as below:

[image]

[image]

[image]

billphipps, 2025-02-26:
Hi Jpgig,

I was just working on updating our instructions and testing to use the later version of ADS with wolfHSM and a TC399, so I'm right in the same place you are.  I'll complete my build and see if I get the same errors that you did.

Just want to confirm a few details of your setup:
Hardware: TC375-LiteKit
Compiler/Build: Windows-based Aurix Developer Studio 1.10.6 (ADS) TriCore GCC 
                      or WSL2 tricore-elf-gcc (assumed Ubuntu-based)

Warmest regards,
Bill

Jpgig, 2025-02-27:
Yes, I'm using TC375-LiteKit
Below is the compiler details:

[image]

[image]

[image]

billphipps, 2025-04-03:
Updated the example project in PR#558 to exclude unused files.

closed 2025-04-03 by billphipps