wolfSSL/wolfBoot #563: wolfBoot build in AURIX studio while importing WSL folder

Jpgig, 2025-04-16:
I followed the instructions as mentioned in the read me file but while importing wolfBoot project directly from the WSL gives compilation error:

[image]

dgarske, 2025-04-16:
Seems like it's missing the wolfCrypt sources files. Did you also get the git submodules?

```
git submodule init
git submodule update
```

Jpgig, 2025-04-16:
Yes I did it, If I copy folder from WSL to windows then it build successfully. But moving files from WSL and windows takes good amount of time. 
Also for last step when I move files from windows to WSL it fails for signature.

bigbrett, 2025-04-16:
I don't believe you can reference files in WSL from windows (where AURIX IDE is running), or at least not without some extra headache/permissions/mounting. I would recommend having all files on your windows host, and accessing them on WSL2 through `/mnt/c`.

Closing because this isn't an issue with wolfBoot.

closed 2025-04-16 by bigbrett

Jpgig, 2025-04-17:
Thanks Brett, I'm able to build as you mentioned above.