wolfSSL/wolfBoot #565: wolfboot with wolfHSM - NVM tool

Jpgig, 2025-04-17:
wbaurixtool.sh script at line no. 331 is using nvmtool to generate the NVM image.
It fails to execute with below error.

[image]

Seems whnvmtool bin is not present. I also searched for instructions to build i but no instructions are available. ANy help to resolve this issue?

Jpgig, 2025-04-21:
I built whnvmtool tool as below:
1) go the folder where whnvmtool is located.
path : \wolfBoot\IDE\AURIX\wolfHSM-infineon-tc3xx\wolfHSM\tools\whnvmtool
execute make command and this will build whnvmtool elf.
2) Give the path of whnvmtool tool in wbaurixtool.sh file
NVMTOOL_DIR="$WOLFBOOT_DIR/IDE/AURIX/wolfHSM-infineon-tc3xx/wolfHSM/tools/whnvmtool"
WHNVMTOOL="$NVMTOOL_DIR/whnvmtool"

then change belwo line 
whnvmtool --image="$NVM_BIN" --size=0x10000 --invert-erased-byte "$NVM_CONFIG"
to "$WHNVMTOOL" --image="$NVM_BIN" --size=0x10000 --invert-erased-byte "$NVM_CONFIG".

This resolved the issue.

bigbrett, 2025-05-15:
fixed in #569

closed 2025-05-15 by bigbrett